NVIDIAGameWorks/PhysX #221: Clarification on Angular Velocity

yaniv-f, 2019-11-12:
Hi. I have a few questions on angular velocity in PhysX:
1. In what units are the values in the PxVec3 returned from PxRigidBody::getAngularVelocity() ?
2. With respect to which coordinate frame is the above angular velocity specified? Is it with respect to the scene (global) coordinate frame or with respect to the (local) rigid body frame ?

Thanks,

 Yaniv

kstorey-nvidia, 2019-11-12:
Units are radians/s and they are returned in global coordinate frame.

yaniv-f, 2019-11-12:
Hi. Thanks for the quick response !

I have a follow-up question: I'm using CARLA (http://carla.org/) to model autonomous vehicles. CARLA uses Unreal + PhysX. I need to calculate the velocity at different parts of a car. Note that when a car is turning, each part will have a different velocity, depending on the distance from the back axle (there will be an additional component which depends on the angular rate of rotation (the angular velocity) and the lever arm from the origin of the rotation which is the center of the back axle). When getting linear velocity from PxRigidBody::getLinearVelocity(), to which point in the rigid body is the velocity associated with? For cars - is it the center of the back axle ? 

Thanks,

 Yaniv

kstorey-nvidia, 2019-11-12:
Velocity is at the COM of the body. The offset between the actor pose (PxRigidBody::getGlobalPose()) and the COM can be found using the PxRigidBody::getCMassLocalPose() method. This offset is in local space. The rotational component relates to the alignment of the inertial frame (so you can ignore this) but the translational part of the CMassLocalPose defines the local space offset to the COM of the body.

I'm sure you are perfectly capable of calculating the velocity at a given location yourself, but there are utility functions in PxRigidBodyExt.h that can compute this for you.

yaniv-f, 2019-11-12:
Hi. Thanks for the pointer! I reviewed the code, and it behaves as expected. I'm closing the issue. I would recommend adding to the API documentation the answers to my first 2 questions.

Thanks and Best Regards,

 Yaniv

closed 2019-11-12 by yaniv-f