NVIDIAGameWorks/PhysX #224: PVD Freezing when clicking on "Profile" tab

Maclodawn, 2019-11-14:
Hello,

When running _PVD_ Version _3.2019.04.26214843_ with _Samples_ project from _PhysXSDK 4.1_ solution on _VS2017_, it successfully connects with PVD and I can see the scene in the _"Render"_ tab, but when I try to go in the _Profile_ tab (most of the time) PVD just freezes.

Is this a normal thing? Did I miss something?
Let me know if you need more information.

Thanks for your help.

FZzzz, 2019-11-29:
I had the same problem one month ago. Although I still didn't figure out how to get into the profile panel via network stream (the one most example use), I found it is possible to use the _Profile_ panel if you export the profiles by file stream. 

To use file stream, please refer to https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/VisualDebugger.html#file-setup

Hope this solves your problem  :=)

amoravanszky, 2019-12-27:
Sorry about this.  We are just starting work on quite a large rewrite of PVD because there are some fundamental problems with the current incarnation that are hard to fix.  So while help is on the way it will still take quite some time to arrive -- I don't have a date so far.

electrocnic, 2023-08-27:
2 more years, bump :D

preist-nvidia, 2023-08-28:
Hi @electrocnic - PVD is no longer supported, so we will not update it or address existing issues. If possible for you, we recommend using PhysX 5 and Omniverse PVD: https://docs.omniverse.nvidia.com/extensions/latest/ext_omnipvd.html

closed 2023-08-28 by preist-nvidia

electrocnic, 2023-08-28:
> Hi @electrocnic - PVD is no longer supported, so we will not update it or address existing issues. If possible for you, we recommend using PhysX 5 and Omniverse PVD: https://docs.omniverse.nvidia.com/extensions/latest/ext_omnipvd.html

Amazing, thanks for the hint, I did not know that, this is really helpful!