NVIDIAGameWorks/PhysX #229: performance issue about setGlobalPose

KaiKai29, 2019-11-30:
Hi,
I'm try to use my own force computer,
so I update(reset position) all actors each step.

These are my setting
1.PxSceneQueryUpdateMode::eBUILD_DISABLED_COMMIT_DISABLED
2.PxScene::sceneQueriesUpdate  after updating all actors' position
3.All actors are built by PxRigidStatic

But I found it's very inefficient, when i have 18 thousands actors.
Do you have any good advice? thanks~

kstorey-nvidia, 2019-12-19:
When you change the position of an actor, it has to invalidate many properties of the actor and also attempt to wake up any actors that were previously touching it to make sure objects aren't left floating in the air. This is generally not the most efficient way to use physx - ideally, you allow the simulation to move the actors.

Have you tried setting forces on physx rigid bodies instead? Alternatively, have you tried kinematic actors and setting kinematic targets?

Having said that, both cases involve you performing write operations on 18000 actors, which is going to be relatively expensive and not parallelizable.

What are you specifically trying to achieve? Have you looked at immediate mode? If you are just trying to do basic simulations, you may find that this interface may be more efficient for your usecase.

KaiKai29, 2019-12-28:
> Have you tried setting forces on physx rigid bodies instead? Alternatively, have you tried kinematic actors and setting kinematic targets?

Actually, I'm not using the mechanics of PhysX.
I just try to find which open-source engine's collision detection is suitable for "discrete element method"

KaiKai29, 2019-12-28:
> When you change the position of an actor, it has to invalidate many properties of the actor and also attempt to wake up any actors that were previously touching it to make sure objects aren't left [...]

Thanks for your reply～
I have already done with it.

closed 2019-12-28 by KaiKai29