NVIDIAGameWorks/PhysX #233: Compile physx4 in a Linux environment

yumianhuli2, 2020-01-02:
How can I compile physx4 in Linux？Have a tutorial?Thank you！

AlesBorovicka, 2020-01-03:
Hi,
 there should be a linux readme file with a list of prerequisites and how to:
[https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/documentation/platformreadme/linux/readme_linux.html](url)

yumianhuli2, 2020-01-07:
> Hi,
> there should be a linux readme file with a list of prerequisites and how to:
> [https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/documentation/platformreadme/linux/readme_linux.html](url)
HI！
I've compiled phyx from the readme, but I want to know where the output is？
For example I want to run samplevehicle in the examples how can I do that?

AlesBorovicka, 2020-01-07:
The binaries should be in bin/linux.clang (release/debug/etc..) directory, run the samples executable and select vehicle sample.

yumianhuli2, 2020-01-07:
> The binaries should be in bin/linux.clang (release/debug/etc..) directory, run the samples executable and select vehicle sample.

ok,I found it in the bin/linux.clang/debug folder and run the vehicle sample.
but how can I run this(samplevehicle) on linux?
[image]

AlesBorovicka, 2020-01-07:
1) source code for snippets is in snippets folder
2) There should be also Samples_64 executable, which you can run and in the app you can select that vehicle sample.

closed 2020-10-17 by yumianhuli2