NVIDIAGameWorks/PhysX #239: Assersion fails in PhysX4.1

uehara-mitsuaki, 2020-01-08:
Hi there,

I'm using PhysX4.1 library and the following Assert occurs frequently.
```
PhysXFoundation_64.dll!`anonymous namespace'::DefaultAssertHandler::operator()(const char * expr, const char * file, int line, bool & ignore) Line 65	C++
PhysX_64.dll!physx::Sq::IncrementalAABBPrunerCore::removeObject(const unsigned int poolIndex, const unsigned int poolRelocatedLastIndex, unsigned int & timeStamp) Line 148	C++
PhysX_64.dll!physx::Sq::ExtendedBucketPruner::removeObject(const physx::Sq::PrunerPayload & object, unsigned int objectIndex, const physx::Sq::PrunerPayload & swapObject, unsigned int swapObjectIndex, unsigned int & timeStamp) Line 378	C++
PhysX_64.dll!physx::Sq::AABBPruner::removeObjects(const unsigned int * handles, unsigned int count) Line 235	C++
PhysX_64.dll!physx::Sq::SceneQueryManager::removePrunerShape(unsigned int compoundId, unsigned __int64 data) Line 375	C++
PhysX_64.dll!physx::NpShapeManager::teardownAllSceneQuery(physx::Sq::SceneQueryManager & sqManager, const physx::PxRigidActor & actor) Line 303	C++
PhysX_64.dll!removeActorT<physx::NpRigidDynamic,physx::Scb::Body>(physx::NpRigidDynamic & actor, physx::Scb::Body & scbActor, physx::NpScene * scene, bool wakeOnLostTouch, bool removeFromAggregate) Line 786	C++
PhysX_64.dll!physx::NpScene::removeRigidDynamic(physx::NpRigidDynamic & body, bool wakeOnLostTouch, bool removeFromAggregate) Line 801	C++
PhysX_64.dll!physx::NpScene::removeActorInternal(physx::PxActor & actor, bool wakeOnLostTouch, bool removeFromAggregate) Line 685	C++
PhysX_64.dll!physx::NpScene::removeActor(physx::PxActor & actor, bool wakeOnLostTouch) Line 667	C++
```

Assertion fails on "PX_ASSERT(foundEntry);" and mAABBTree[0].tree, mAABBTree[1].tree are pointing to Null.
```
bool IncrementalAABBPrunerCore::removeObject(const PoolIndex poolIndex, const PoolIndex poolRelocatedLastIndex, PxU32& timeStamp)
{
	// erase the entry and get the data
	IncrementalPrunerMap::Entry entry;
	bool foundEntry = true;
	const PxU32 treeIndex = mAABBTree[mLastTree].mapping.erase(poolIndex, entry) ? mLastTree : mCurrentTree;
	// if it was not found in the last tree look at the current tree
	if(treeIndex == mCurrentTree)
		foundEntry = mAABBTree[mCurrentTree].mapping.erase(poolIndex, entry);	// exit somethings is wrong here, entry was not found here
	PX_ASSERT(foundEntry);
	if(!foundEntry)
		return false;	// tree must exist
	PX_ASSERT(mAABBTree[treeIndex].tree);
	CoreTree& tree = mAABBTree[treeIndex];
	timeStamp = tree.timeStamp;
```

Can anyone help me solve this problem?

Thanks.

AlesBorovicka, 2020-01-08:
Hi,
 please do you have a repro for this issue? Or any detailed steps when this issue might happen? If both trees are null, that should mean, that no additional object was inserted into scene, then strictly speaking the removeObject should not happen.
Workaround for this issue might be to switch to bucket pruner, you can change:
#define USE_INCREMENTAL_PRUNER 1 // change to -> 0 in SqExtendedBucketPruner.h
But the problem might still remain, if the entry was not found and the trees are both null, then something went wrong in the SqAABBPruner most likely. 
You could also comment out the assert and early exit as the code already does, but I would be interested to know when this situation can happen.

uehara-mitsuaki, 2020-01-08:
Thanks for your reply, Ales.
I can repro this, so I try your advice.

uehara-mitsuaki, 2020-01-14:
I changed USE_INCREMENTAL_PRUNER to 0, but assertion failed again in "updateObject" function.

I was using binary serialized data but if I use original data and create actors by PxCreateStatic and createHeightField instead, it works.

https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/Serialization.html

There might be some issues when serializing or loading data.

AlesBorovicka, 2020-01-15:
Thanks for the additional info, we have some internal tests using scene queries and serialization, but it looks like there is some issue. 
I can try to repro on our side. So far I guess what might do is:
 * load a scene through the serialization feature
 * call directly a remove on some object maybe, so that the internal trees are not yet even created as they were not used? Could it be?

uehara-mitsuaki, 2020-01-16:
> call directly a remove on some object maybe, so that the internal trees are not yet even created as they were not used? Could it be?

I call PxScene::removeActor when I remove dynamic actor.
Is it correct?

AlesBorovicka, 2020-01-16:
Yes, that should be correct. Not sure what is going on, there seems to be nothing suspicious. Will try to reproduce the issue.
Thanks for the detailed information.

uehara-mitsuaki, 2020-01-22:
I set breakpoints inside AABBPruner class in SqAABBPruner.cpp and insdie IncrementalAABBPrunerCore class in SqIncrementalAABBPrunerCore.cpp with version 4.1.0 and this is what happens following above steps from 7th to 10th in detail.

7. remove the CharacterActor by removeActor()
All active actors get removed by removeActor(), and the two methods are called.
AABBPruner::release() -> AABBPruner::mAABBTree becomes null.
IncrementalAABBPrunerCore::release() -> trees in IncrementalAABBPrunerCore::mAABBTree become null.

8. add a dynamic actor named CharacterActor2
9. add a dynamic actor named BulletActor2
I call PxScene::addActor() but AABBPruner::mAABBTree is null, so 
IncrementalAABBPrunerCore::addObject() will not be called and trees are still null.

10. call sweep() and setKinematicTarget() for BulletActor2 every frame.
AABBPruner::fullRebuildAABBTree() will be called by sweep(), and AABBPruner::mAABBTree will be created.

after this, IncrementalAABBPrunerCore::updateObject() will be called but trees are still null, so the assertion will fail.

Now I'm trying to create a source code that reproduces this, but I'm having a hard time.
I'll post it if I can create one.

AlesBorovicka, 2020-01-22:
Right, what you traced is how it should behave. If you delete all objects from scene queries the trees are released. 
Now when you start adding new objects, nothing happens until a scene query happens. The fullRebuildAABBTree should build the tree for SqAABBPruner. So far so good. 
But here I dont understand how you get IncrementalAABBPrunerCore::updateObject() call.
In the SqAABBPruner, there should be code like this:
`			const PoolIndex poolIndex = mPool.getIndex(handles[i]);
			const TreeNodeIndex treeNodeIndex = mTreeMap[poolIndex];
			if(treeNodeIndex!=INVALID_NODE_ID) // this means it's in the current tree still and hasn't been removed
				mAABBTree->markNodeForRefit(treeNodeIndex);
			else // otherwise it means it should be in the bucket pruner
			{
				bool found = mBucketPruner.updateObject(newBounds[poolIndex], payloads[poolIndex], poolIndex);
				PX_UNUSED(found); PX_ASSERT(found);
			}
`
This means, that the bucketpruner (incrementalAABBpruner) gets called if the object was not found in the main AABBTree. There should be sq mutex guarding for concurrent read/write, so I have no idea how this can happen. You should get a call into the bucket pruner update only if the object was not found in the main aabbtree. But since your sweep call did recomputed that tree, it should contain all objects. 

It would be really great to have a repro for this.