NVIDIAGameWorks/PhysX #240: PhysX Plugin for 3ds Max 2016 issue

Andre171, 2020-01-11:
I have a big issue with the physX Plugin. I have a Character created and the shape is little more detailed. I have tried to make a custom mesh for the kinemetric option and the cloth still goes through the character. Am I missing something, is the cloth plugin not working, why does the plugin have a custom mesh but doesn't work for clothing, I don't know. But I really need to get things going with my game.
[image]

GeneralSoundwave, 2020-02-07:
You are one of the few people that I have see that is using max that actually extracted some kind of exporter from this Repo. Where on earth did you find the install?

I would love to even get to the stage where youre at! I thought we simply export the character with the fabric directly into unreal and handle the cloth on that side... not on the max-side. I'm using 2017 btw, so the issues shouldnt be dissimilar. But i need to get to the point where you are. Can you help my friend and i? Maybe we can help eachother.

Here is what i know on how to install this thing.

It's a shame that I've been looking for stuff for almost 2 weeks and there's only dated install info on what to use. It's difficult to find what you're looking for if the vocabulary isn't taught. nearly 20 years doing this and after all this time, programmers still don't know how to talk to artists.

My friend and I were able to find this link to the newest Repo to PhsX 4.1.1:
https://github.com/NVIDIAGameWorks/PhysX

We downloaded the Repo and behold... more programmer language that needs deciphering:

[CODE]## Quick Start Instructions

Requirements:
* Python 2.7.6 or later
* CMake 3.12 or later

To begin, clone this repository onto your local drive.
Then change directory to physx/, run ./generate_projects.[bat|sh] and follow on-screen prompts.
This will let you select a platform specific solution to build.
You can then open the generated solution file with your IDE and kick off one or more configuration builds.[/CODE]

Here's what I understand... Typically, when normal people install things, we use an *.EXE. There are some in the directory (see attached). However, in the README file, it doesnt state which one of these will work with Max. 

So if there is no exe, that means we have to compile our own stuff. 

I found this Python Compiler. My instinct is telling me that this is a stand alone piece of software that has absolutely nothing to do with obtaining a plugin for Max.
https://www.onlinegdb.com/online_python_compiler

Hmmm... conflicting info much? I will be posting my frustrations inside the NVIDIA and Max forums as well.

Andre171, 2020-02-12:
Good day.

1st of. my name is Andre and I am in South africa.
2nd. I have almost 12 Years of experience in 3ds Max.

Now to the point.

I have found the installation for 3DS Max and found that there are two
different files for the physx plugin
try this link and it should work. https://developer.nvidia.com/physx-sdk
and installed the dot net frameworks 4.5 to install the plugin.

1st:  I created the character first, But also I created the character to
let me extract the clothing from the character
In other words the character is also a template for clothes. This makes it
much easier and less hassles on clothes creating clippings (going through)
the character when animated.

2nd: When I have finished the skinning process of my character. I save the
skinning information in a folder where I organize my characters, buildings,
foliage and small props. Then I import my skinning information on the
clothes, Both will have the same skinning information.

I am extremely serious at what I do, for the sole purpose of income.

Andre171, 2020-02-12:
Oh I forgot. I am using 3ds Max 2016