NVIDIAGameWorks/PhysX #243: Limits for Revolute Joint bigger than +- Pi doesn't work

xxxdomi, 2020-01-20:
Hello,

in my code, i'm creating a hinge constraint (for a revolute joint) like that...

PxRevoluteJoint pxconstraint = pxenv->CreateAndAddHingeConstraint(actor0, &frame0, actor1, &frame1);

pxconstraint->setLimit(PxJointAngularLimitPair(constraint->LowerLimit, constraint->UpperLimit, 0.01f));
pxconstraint->setRevoluteJointFlag(PxRevoluteJointFlag::eLIMIT_ENABLED, true);
	
If I set limits between [-Pi; +Pi] everything works fine and my object rotates until the limits. If I use limits which are bigger, for example [-2*Pi*0.75; 2*Pi*0.75], it doesn't work anymore. It seems, that just angles between -180 and 180 degrees work. 

In my application I would like limit the angles in the full range between -360 and +360 degrees, like descripted in the "setLimit" function. Do you have any ideas, if I did something wrong?

Best regards,

Dominik

kstorey-nvidia, 2020-01-20:
You didn't mention which version of PhysX you are using. For quite a while, PhysX only supported limits within the range +/- Pi. More recently, it was extended to support wider limit ranges. This had some knock-on effects for legacy users upgrading to newer versions of PhysX. If their joint frames were not in the same hemisphere, PhysX would have previously chosen the shortest path but now it would not, potentially causing behavioral regressions. As a result, wider limit ranges were disabled by default and can be enabled with the flag:

PxConstraintFlag::eENABLE_EXTENDED_LIMITS

Hope this helps

xxxdomi, 2020-01-20:
I'm using PhysX 4.1.

That work's for me, thank you!

closed 2020-01-20 by xxxdomi