NVIDIAGameWorks/PhysX #246: Problems simulating 'long' objects in PhysX

davidstark481, 2020-01-30:
Hello,

I am using PhysX 4.1 to simulate around 300 dynamic rigid bodies (each of which contain around 30-40 shapes coming from a convex hull decomposition).  

With most objects, I get good results, but when my object is 'long' i.e. the object is significantly longer in one dimension than in the other two (eg. 2000mm x 20mm x 20mm), I end up with a lot of problems.  PhysX typically tends to freeze after so many time steps, and the time steps are very slow.  There is also lots of penetration between the objects (which may be causing this).

My parameters are very 'careful'.  The center of inertia is reasonable; it is within the object.  I have the solver iteration counts set to 255 and the MaxDepenetrationVelocity set to 3.0.  The density of each object is set to 0.1.  I am running the simulation with a small time step size of 1/2000, so as to minimize any penetration as much as possible.
The scene flags set are: eENABLE_GPU_DYNAMICS, eENABLE_PCM, eGPU.

Any help or suggestions would be appreciated.  I can provide other information if required.

Thanks
Dave

kstorey-nvidia, 2020-01-30:
Can you provide a repro or a PVD capture of these issues?

davidstark481, 2020-02-03:
You can find a PVD capture at the following link:

https://www.sendthisfile.com/3FY5xixGrhWszVEXlRG2TfcF

In this case, the simulation was very slow and ended up freezing.

Thanks,
Dave

kstorey-nvidia, 2020-02-03:
Have you checked to see if there are any errors emitted from PhysX?

Looking at the settings in your scene, it seems like the following may be the case:

(1) You have a pile of ~800 rigid bodies and a total of 22400 convex meshes
(2) You have a contact offset of 0.8. You seem to be using meter units, which means that you will generate contacts between a pair of shapes if they are within 1.6m of each other. This will generate a huge number of contacts.
(3) You are using the default settings for GPUDynamicsConfig.
(4) You do not appear to have set up inertia tensors correctly. You have bodies with mass of 0.00005 and inertia tensors of (1,1,1). This will not behave correctly and will be extremely difficult for the solver to handle, especially given that they are long/thin.

My expectation is that you are overflowing the contact/constraint buffer capacity defined in the GPUDynamicsConfig. This should trigger error messages. Ideally, it wouldn't cause a crash/hang but there must be some case that we didn't catch adequately in the solver.

My suggestion to go forwards would be as follows:

(1) Reduce the contact offset. A very large contact offset will generate a huge number of unnecessary contacts that will damage performance and mean you need to pre-allocate very large contact buffers. Try reducing this to something closer to the default (0.02). Given that you are running at small time-steps, this should be sufficient.
(2) Try to increase the buffer sizes in the GPUDynamicsConfig. Try doubling or maybe quadrupling them.
(3) Compute appropriate inertia tensors.
(4) Reconsider 255 solver iterations. You shouldn't need that many for stacks of a few 100 bodies, especially if running at very small time-steps.
(5) Try increasing maxGpuPartitions in the sceneDesc to 32 from its default 8 value. Note that 32 is the maximum value it can take and it can only be a power-of-two.

Additionally, it looks like all the contact generation is running exclusively on the CPU. This probably suggests that you didn't cook your convex hulls with the correct flags to generate GPU data. Please ensure this is enabled because GPU contact gen is much faster and also avoids DMAs from host to device.

kstorey-nvidia, 2020-02-03:
Also, we just noticed that you are running debug build in this capture. I'd recommend running checked or profile build for PVD captures. This should give us more reasonable understanding of the performance characteristics of the scene. Debug build will be a lot slower than an optimized build

davidstark481, 2020-02-06:
Ok Thanks.

These improvements have made the simulation a lot more stable.  I hadn't even considered the distribution of mass.  It is also a little bit faster too.

Dave

closed 2020-02-06 by davidstark481