NVIDIAGameWorks/PhysX #248: How to install PhysX for 3D Max... For Artists

GeneralSoundwave, 2020-02-07:
I'm going to keep this post live, as long as it takes, until a solution presents itself. It's a shame that I've been looking for stuff for almost 2 weeks and there's only dated install info on what to use. It's difficult to find what you're looking for if the vocabulary isn't taught. nearly 20 years doing this and after all this time, programmers still don't know how to talk to artists.

My friend and I were able to find this link to the newest Repo to PhsX 4.1.1:
https://github.com/NVIDIAGameWorks/PhysX

We downloaded the Repo and behold... more programmer language that needs deciphering:

[CODE]## Quick Start Instructions

Requirements:
* Python 2.7.6 or later
* CMake 3.12 or later

To begin, clone this repository onto your local drive.
Then change directory to physx/, run ./generate_projects.[bat|sh] and follow on-screen prompts.
This will let you select a platform specific solution to build.
You can then open the generated solution file with your IDE and kick off one or more configuration builds.[/CODE]

Here's what I understand... Typically, when normal people install things, we use an *.EXE. There are some in the directory (see attached). However, in the README file, it doesnt state which one of these will work with Max. 

So if there is no exe, that means we have to compile our own stuff. 

I found this Python Compiler. My instinct is telling me that this is a stand alone piece of software that has absolutely nothing to do with obtaining a plugin for Max.
https://www.onlinegdb.com/online_python_compiler

Hmmm... conflicting info much? I will be posting my frustrations inside the NVIDIA and Max forums as well. 
[image]

mottosso, 2020-02-20:
Hi @GeneralSoundwave, spotted this after having replied to your post on the NVIDIA forums, here.

- https://devtalk.nvidia.com/default/topic/1069297/physics-modeling/starting-to-work-with-physx/

I might add that there are links to the 3ds Max integration here.

- https://www.nvidia.com/en-us/drivers/physx-dcc-plugins/

Although those are quite dated, and I'm not sure there is a more recent integration of PhysX into 3ds. That's about as proper solution I think you can hope to find, unfortunately.

It's not immediately obvious, but PhysX integration and PhysX library are different projects, and not necessarily something provided by the same people. This issue section, along with the other forum linked to above, are unrelated to DCCs like 3ds and game engines Unity and Unreal for example. For those, it's probably better to visit the forums for the DCCs instead, as they would be the ones maintaining the integration of PhysX into their software. 

For PhysX staff, since this is where people arrive from Googling PhysX, could it be worth collecting some links in the README, to direct newcomers into the right direction ? I'd imagine programmers come here, but also artists such as the gentleman above, and likely also gamers interested in knowing more about the tech behind the ragdolls in their favorite games.

Something along the lines of..

| Project | Description
|:-------------------|:-----------
| **PhysX SDK** | A C++ library for rigid body simulation (You Are Here)
| **PhysX Integrations** | With links to Unity and Unreal forums, and other engines too perhaps, along with maybe the outdated DCC plug-ins for posterity, and to underline that there isn't anything newer. At least not from NVIDIA.
| **PhysX Technology** | Along with the logo that appears in games. With maybe a high-level description of what PhysX means to the casual gamer and perhaps an entry point for those diving into programming or games development for the first time.

I suspect that could help clear up some confusion, and keep issues such as this one from appearing in the wrong spot, leaving both poster and developers frustrated about not being able to proceed.

ForgeSouth, 2020-09-03:
@GeneralSoundwave Thank you so much for this! I have had exactly the same frustration as you!!

@mottosso Thank you for the reply! I finally have an .exe to install!!!