NVIDIAGameWorks/PhysX #249: How in include Physx in a cmake project

arijeetbaruah, 2020-02-17:
I am using PhysX and cmake to make a game for my university project. However, I am having issues with cmake library. My PhysX code is added as a git submodule within my project in submodules directory
```
cmake_minimum_required(VERSION 3.14)
## Giving a name to the project
project(weird_golf_game)

#Setup CMake to run tests
enable_testing()

## Including headers, from both the current project and from the imported libraries
include_directories(include)
...
include_directories(submodules/PhysX/physx/include)
add_subdirectory(submodules/PhysX/physx/compiler/public)

## I want to use the C++17 standard, forsooth!
set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_STANDARD_REQUIRED ON)

# OpenGL
find_package(OpenGL REQUIRED)

add_executable(weird_golf_game ...)
target_link_libraries(weird_golf_game ... opengl32 PhysX)
```

but on running cmake I am getting these error:
```
PHYSX ROOT D:/Source/Repos/weird-golf-game/submodules/PhysX/physx
PhysX Build Platform: 
Using CXX Compiler: C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/VC/Tools/MSVC/14.23.28105/bin/Hostx64/x64/cl.exe
CMake Error at submodules/PhysX/externals/cmakemodules/NvidiaBuildOptions.cmake:48 (MESSAGE):
  When using the GameWorks output structure you must specify
  PX_OUTPUT_LIB_DIR as the base
Call Stack (most recent call first):
  submodules/PhysX/physx/source/compiler/cmake/CMakeLists.txt:70 (INCLUDE)

Configuring incomplete, errors occurred!
See also "D:/Source/Repos/weird-golf-game/out/CMakeFiles/CMakeOutput.log".
```

dwhinham, 2020-02-17:
Hi Arijeet,

I've recently used PhysX 4 with CMake in this way, although it's probably not an intended/supported use.

The proposed changes in #222 and #208 will probably make using PhysX in our own projects easier in the future.

Until then, try something like this just before your `add_subdirectory()` line:
```
# PhysX-specific CMake project setup
set(NV_USE_DEBUG_WINCRT ON CACHE BOOL "Use the debug version of the CRT")
set(PHYSX_ROOT_DIR ${CMAKE_SOURCE_DIR}/externals/physx/physx)
set(PXSHARED_PATH ${PHYSX_ROOT_DIR}/../pxshared)
set(PXSHARED_INSTALL_PREFIX ${CMAKE_INSTALL_PREFIX})
set(CMAKEMODULES_VERSION "1.27")
set(CMAKEMODULES_PATH ${PHYSX_ROOT_DIR}/../externals/cmakemodules)
set(PX_OUTPUT_LIB_DIR ${CMAKE_CURRENT_BINARY_DIR}/externals/physx)
set(PX_OUTPUT_BIN_DIR ${CMAKE_CURRENT_BINARY_DIR}/externals/physx)

# Call into PhysX's CMake scripts
add_subdirectory(${PHYSX_ROOT_DIR}/compiler/public externals/physx)
```

arijeetbaruah, 2020-02-18:
It's working now, thankx

closed 2020-02-18 by arijeetbaruah

shehabattia96, 2020-11-04:
Here towards the end of 2020 and #222 , #208 are still open. It appears on 4.1, cmake_generate_projects requires some more variables: CMAKE_CONFIGURATION_TYPES  and TARGET_BUILD_PLATFORM 

```
#### Build PhysX library ####
# PHYSX_PATH - path to the `{cloned repository}/physx` repo directory git://github.com/NVIDIAGameWorks/PhysX.git
set( PHYSX_ROOT_DIR $ENV{PHYSX_PATH} ) #This is needed for $ENV{PHYSX_PATH}/compiler/public/CMakeLists.txt
set( PHYSX_INCLUDE_DIRS $ENV{PHYSX_PATH}/include/ $ENV{PHYSX_PATH}/../pxshared/include/ )
set( PHYSX_LIBRARIES
	PhysXExtensions
	PhysX
	PhysXPvdSDK
	PhysXVehicle
	PhysXCharacterKinematic
	PhysXCooking
	PhysXCommon
	PhysXFoundation
	# SnippetUtils
)

set(CMAKE_CONFIGURATION_TYPES "debug") #enum configTypes {debug checked profile release}
set(TARGET_BUILD_PLATFORM "windows") # has to match the TARGET_BUILD_PLATFORM in $ENV{PHYSX_PATH}/physix/buildtools/cmake_generate_projects.py
set(PX_BUILDSNIPPETS OFF CACHE BOOL "Generate the snippets")
set(PX_BUILDPUBLICSAMPLES OFF CACHE BOOL "Generate the samples projects")
set(PX_GENERATE_STATIC_LIBRARIES OFF CACHE BOOL "Generate static libraries")
set(PX_FLOAT_POINT_PRECISE_MATH OFF CACHE BOOL "Float point precise math")
set(NV_USE_STATIC_WINCRT ON CACHE BOOL "Use the statically linked windows CRT")
set(NV_USE_DEBUG_WINCRT ON CACHE BOOL "Use the debug version of the CRT")
set(PXSHARED_PATH $ENV{PHYSX_PATH}/../pxshared)
set(PXSHARED_INSTALL_PREFIX ${CMAKE_INSTALL_PREFIX})
set(CMAKEMODULES_VERSION "1.27")
set(CMAKEMODULES_PATH $ENV{PHYSX_PATH}/../externals/cmakemodules)
set(PX_OUTPUT_LIB_DIR ${CMAKE_CURRENT_BINARY_DIR}/externals/physx)
set(PX_OUTPUT_BIN_DIR ${CMAKE_CURRENT_BINARY_DIR}/externals/physx)
  
# Call into PhysX's CMake scripts
add_subdirectory($ENV{PHYSX_PATH}/compiler/public externals/physx)

# Add physx libraries to target
target_link_libraries(target_name PUBLIC ${PHYSX_LIBRARIES})
```

blueage77, 2020-11-24:
Hi thanks for that. But I now am getting C2220 and C5045 compiler errors and cannot solve them. Any suggestions? Google did not help.

shehabattia96, 2020-11-25:
@cemoezdemir please share your CMakeLists file, path to PhysX and informative lines from the build log. Thanks.

cemoezdemir, 2020-11-25:
`cmake_minimum_required(VERSION 3.18)
project(game_lab_3)

set(CMAKE_CXX_STANDARD 17)

#### Build PhysX library ####
# PHYSX_PATH - path to the `{cloned repository}/physx` repo directory git://github.com/NVIDIAGameWorks/PhysX.git
set( ENV{PHYSX_PATH} C:/dev/coezdemir/GL3/05-gl3-cem-ozdemir/code/extern/physx/physx)
set( PHYSX_ROOT_DIR $ENV{PHYSX_PATH} ) #This is needed for $ENV{PHYSX_PATH}/compiler/public/CMakeLists.txt
set( PHYSX_INCLUDE_DIRS $ENV{PHYSX_PATH}/include/ $ENV{PHYSX_PATH}/../pxshared/include/ )
set( PHYSX_LIBRARIES
        PhysXExtensions
        PhysX
        PhysXPvdSDK
        PhysXVehicle
        PhysXCharacterKinematic
        PhysXCooking
        PhysXCommon
        PhysXFoundation
        # SnippetUtils
        )

set(CMAKE_CONFIGURATION_TYPES "debug") #enum configTypes {debug checked profile release}
set(TARGET_BUILD_PLATFORM "windows") # has to match the TARGET_BUILD_PLATFORM in $ENV{PHYSX_PATH}/physix/buildtools/cmake_generate_projects.py
set(PX_BUILDSNIPPETS OFF CACHE BOOL "Generate the snippets")
set(PX_BUILDPUBLICSAMPLES OFF CACHE BOOL "Generate the samples projects")
set(PX_GENERATE_STATIC_LIBRARIES OFF CACHE BOOL "Generate static libraries")
set(PX_FLOAT_POINT_PRECISE_MATH OFF CACHE BOOL "Float point precise math")
set(NV_USE_STATIC_WINCRT ON CACHE BOOL "Use the statically linked windows CRT")
set(NV_USE_DEBUG_WINCRT ON CACHE BOOL "Use the debug version of the CRT")
set(PXSHARED_PATH $ENV{PHYSX_PATH}/../pxshared)
set(PXSHARED_INSTALL_PREFIX ${CMAKE_INSTALL_PREFIX})
set(CMAKEMODULES_VERSION "1.27")
set(CMAKEMODULES_PATH $ENV{PHYSX_PATH}/../externals/cmakemodules)
set(PX_OUTPUT_LIB_DIR ${CMAKE_CURRENT_BINARY_DIR}/extern/physx)
set(PX_OUTPUT_BIN_DIR ${CMAKE_CURRENT_BINARY_DIR}/extern/physx)

# Call into PhysX's CMake scripts
add_subdirectory($ENV{PHYSX_PATH}/compiler/public extern/physx)
add_subdirectory(extern/glad)
add_subdirectory(extern/glfw)
add_subdirectory(extern/glm)
add_subdirectory(extern/soloud)
add_subdirectory(extern/tinyobj)

add_subdirectory(game)

You can see the path above. If something seems out of place I am new to CMake. 

And this is my second CMakeList:
`cmake_minimum_required(VERSION 3.18)

# set variables for source files
set(SOURCE_FILES src/main.cpp src/rendering/Shader.h src/rendering/Shader.cpp src/rendering/meshes/Mesh.cpp src/rendering/meshes/Mesh.h src/rendering/meshes/Cube.cpp src/rendering/meshes/Cube.h src/rendering/meshes/Sphere.cpp src/rendering/meshes/Sphere.h  src/rendering/scenegraph/visitors/Visitor.h src/rendering/scenegraph/nodes/Node.h src/rendering/scenegraph/nodes/GroupNode.cpp src/rendering/scenegraph/nodes/GroupNode.h src/rendering/scenegraph/nodes/SphereNode.cpp src/rendering/scenegraph/nodes/SphereNode.h src/rendering/scenegraph/nodes/CubeNode.cpp src/rendering/scenegraph/nodes/CubeNode.h src/rendering/scenegraph/visitors/RasterVisitor.cpp src/rendering/scenegraph/visitors/RasterVisitor.h src/rendering/scenegraph/nodes/CameraNode.h src/rendering/scenegraph/nodes/CameraNode.cpp src/rendering/scenegraph/nodes/LightNode.cpp src/rendering/scenegraph/nodes/LightNode.h src/rendering/meshes/OBJMesh.cpp src/rendering/meshes/OBJMesh.h src/rendering/scenegraph/nodes/OBJNode.cpp src/rendering/scenegraph/nodes/OBJNode.h src/actors/Actor.cpp src/actors/Actor.h src/actors/Car.cpp src/actors/Car.h)

# set executable name
set(EXE_FILE Game)

# add the executable target
add_executable(${EXE_FILE} ${SOURCE_FILES})

# require C++ 17 compiler
target_compile_features(${EXE_FILE} PRIVATE cxx_std_17)

# link with dependencies
target_link_libraries(${EXE_FILE} PRIVATE glad glfw glm soloud tinyobj ${PHYSX_LIBRARIES})

target_compile_definitions(${EXE_FILE} PRIVATE DEBUG_ASSET_ROOT=${PROJECT_SOURCE_DIR})
`

The CMake build finishes without problems, but then I get this when compiling:

`====================[ Build | Game | Debug ]====================================
C:\dev\coezdemir\cmake-3.18.4-win64-x64\bin\cmake.exe --build C:\dev\coezdemir\GL3\05-gl3-cem-ozdemir\code\cmake-build-debug --target Game
[  0%] Building CXX object extern/physx/sdk_source_bin/CMakeFiles/PhysXFoundation.dir/__/__/foundation/src/PsAllocator.cpp.obj
PsAllocator.cpp
c:\dev\coezdemir\gl3\05-gl3-cem-ozdemir\code\extern\physx\physx\source\foundation\include\pshashinternals.h(405) : error C2220: Warnung wird als Fehler interpretiert, es wurde keine object-Datei generiert.
c:\dev\coezdemir\gl3\05-gl3-cem-ozdemir\code\extern\physx\physx\source\foundation\include\pshashinternals.h(405) : warning C5045: Der Compiler f�gt die Spectre-Entsch„rfung f�r die Arbeitsspeicherauslastung ein, wenn die Option /Qspectre angegeben wird.
c:\dev\coezdemir\gl3\05-gl3-cem-ozdemir\code\extern\physx\physx\source\foundation\include\pshashinternals.h(403) : Hinweis: Der Bereich des Index "bucket" wird durch Vergleich in dieser Zeile gepr�ft.
c:\dev\coezdemir\gl3\05-gl3-cem-ozdemir\code\extern\physx\physx\source\foundation\include\pshashinternals.h(417) : Hinweis: F�gt Arbeitsspeicherauslastung in dieser Zeile hinzu.
NMAKE : fatal error U1077: "C:\PROGRA~2\MICROS~1\2017\COMMUN~1\VC\Tools\MSVC\1416~1.270\bin\Hostx86\x86\cl.exe": Rückgabe-Code "0x2"
Stop.
NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\nmake.exe"": Rückgabe-Code "0x2"
Stop.
NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\nmake.exe"": Rückgabe-Code "0x2"
Stop.
NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\nmake.exe"": Rückgabe-Code "0x2"
Stop.`

shehabattia96, 2020-11-25:
I do not see anything in the logs that may help, except "NMAKE: fatal error U1077", which when I googled seems to be a problem with the Visual Studio installation or a missing path to compiler executable.

I suspect you've been using your environment for a while, so I'm not really sure what's wrong with nmake. However, I personally use CMake (cmake.org) and I've had zero problems, if it's not too much trouble could you give it a try? Thanks.

cemoezdemir, 2020-11-25:
Thanks for you fast reply, but I am not sure if I am understanding you right. I am using CMake already, version 3.18.4 to be exact. Maybe I also should mention, that I am using CLion.

shehabattia96, 2020-11-25:
Your logs say you're running NMAKE. The issue is "NMAKE: fatal error U1077". I was suggesting to fix nmake, I suspect you should go into your Build Configuration and choose the path to the compiler: https://www.jetbrains.com/help/clion/clion-quick-start-guide.html#build-and-run.

The alternative I was suggesting is to quickly install cmake (cmake.org), clean, configure, generate and run `cmake --build /path/to/build/folder/` from a command line to see if the problem would go away.

cemoezdemir, 2020-11-25:
Hi again. I managed to get it to work. I just needed PhysX 4.1 and added some ccompiler options to stop treating warnings as errors. Thank you for your help!