NVIDIAGameWorks/PhysX #252: Failing to run generate_projects.bat

piluve, 2020-02-22:
Running `generate_projects.bat` generates this error: 
```
exit /b 0
Hit ENTER to continue...
```
**Branch:** 4.1
**CMake version:**  3.17.0-rc1
**Python version:** 3.6.2 (v3.6.2:5fd33b5, Jul  8 2017, 04:14:34)

I´m not getting any prompts to select a preset.
Any help would be appreciated.
Thanks.

piluve, 2020-02-25:
I had to change this line:

`%PM_PYTHON% "%PHYSX_ROOT_DIR%/buildtools/cmake_generate_projects.py" %1`

to this:

`py "%PHYSX_ROOT_DIR%/buildtools/cmake_generate_projects.py" %1`

`%PM_PYTHON%` was evaluating to python.exe, so not quite sure why it wasn't working. Anyways, I can work with this. Closing.

closed 2020-02-25 by piluve

lyonb96, 2020-03-01:
Just had the same issue, python.exe doesn't work as a command but py does, and this causes the script to silently fail. Anyway, thanks for the work around, hopefully they'll resolve this because it's very annoying that a CMake script, which is made to avoid such annoying issues with C++, would require tweaking out of the box.