NVIDIAGameWorks/PhysX #253: How to build for PS4

arijeetbaruah, 2020-02-24:
We are trying to build the project for both Windows and PS4. We have the PS4 SDK and ORBIS platform but can't find any `.a` files. We are using it for our University project

Nanoatic, 2020-02-25:
```
<?xml version="1.0" encoding="utf-8"?>
<preset name="ps4Orbis" comment="ps4 orbis PhysX SDK general settings">
  <platform targetPlatform="ps4" compiler="clang" />
  <CMakeSwitches>
    <cmakeSwitch name="PX_BUILDSNIPPETS" value="True" comment="Generate the snippets" />
    <cmakeSwitch name="PX_BUILDPUBLICSAMPLES" value="True" comment="Generate the samples" />
    <cmakeSwitch name="PX_GENERATE_STATIC_LIBRARIES" value="True" comment="Generate static libs" />
  </CMakeSwitches>
  <CMakeParams>
    <cmakeParam name="CMAKE_INSTALL_PREFIX" value="install/ps4Orbis/PhysX" comment="Install path relative to PhysX SDK root" />
  </CMakeParams>
</preset>
```
try adding that preset to physx/buildtools/presets/ , name it ps4orbis.xml, and in the root folder go to externals/cmakemodules/ and add "ps4" folder and create inside it a "PS4Toolchain.txt" file and fill it with the following :
```
if( _PS4_TOOLCHAIN_ )
	return()
endif()
set(_PS4_TOOLCHAIN_ 1)

# PS4 SCE version requirement
set(REQUIRED_PS4_VERSION "4.000")

# Get PS4 SCE environment
if( EXISTS "$ENV{SCE_ROOT_DIR}" AND IS_DIRECTORY "$ENV{SCE_ROOT_DIR}" )
	string(REGEX REPLACE "\\\\" "/" PS4_ROOT $ENV{SCE_ROOT_DIR})
	string(REGEX REPLACE "//" "/" PS4_ROOT ${PS4_ROOT})
	if( EXISTS "$ENV{SCE_ORBIS_SDK_DIR}" AND IS_DIRECTORY "$ENV{SCE_ORBIS_SDK_DIR}" )
		string(REGEX REPLACE "\\\\" "/" PS4_SDK $ENV{SCE_ORBIS_SDK_DIR})
		string(REGEX REPLACE "//" "/" PS4_SDK ${PS4_SDK})
		get_filename_component(SCE_VERSION "${PS4_SDK}" NAME)
	endif()
endif()

# Report and check version if it exist
if( NOT "${SCE_VERSION}" STREQUAL "" )
	message(STATUS "PS4 SCE version found: ${SCE_VERSION}")
	if( NOT "${SCE_VERSION}" MATCHES "${REQUIRED_PS4_VERSION}+" )
		message(WARNING "Expected PS4 SCE version: ${REQUIRED_PS4_VERSION}")
		if( PLATFORM_TOOLCHAIN_ENVIRONMENT_ONLY )
			set(PS4_ROOT)
			set(PS4_SDK)
		endif()
	endif()
endif()

# If we only want the environment values, exit now
if( PLATFORM_TOOLCHAIN_ENVIRONMENT_ONLY )
	return()
endif()

# We are building PS4 platform, fail if PS4 SCE not found
if( NOT PS4_ROOT OR NOT PS4_SDK )
	message(FATAL_ERROR "Engine requires PS4 SCE SDK to be installed in order to build PS4 platform.")
endif()

# Tell CMake we are cross-compiling to PS4 (Orbis)
set(CMAKE_SYSTEM_NAME Orbis)
set(PS4 True)

# Set CMake system root search path
set(CMAKE_SYSROOT "${PS4_ROOT}")

# Set compilers to the ones found in PS4 SCE SDK directory
set(CMAKE_C_COMPILER "${PS4_SDK}/host_tools/bin/orbis-clang.exe")
set(CMAKE_CXX_COMPILER "${PS4_SDK}/host_tools/bin/orbis-clang++.exe")
set(CMAKE_ASM_COMPILER "${PS4_SDK}/host_tools/bin/orbis-as.exe")

# Only search the PS4 SCE SDK, not the remainder of the host file system
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_PACKAGE ONLY)

```
try tweaking one the files if you get some errors.
best of luck.

amoravanszky, 2020-02-25:
Nanoatic, I deleted your comment because I believe it violates a console vendor NDA.  Please PM me if you disagree.

amoravanszky, 2020-02-25:
arijeetbaruah, PS4 SDK relevant information is not included in this github and cannot be discussed here in the open as it is subject to an NDA with Sony.  If you are a licensed PS4 developer please email physxlicensing@nvidia.com for access to PhysX on PS4.

closed 2020-02-25 by amoravanszky