NVIDIAGameWorks/PhysX #255: Trigger event fired by PxRigidDynamic::setKinematicTarget() is one frame later.

sugano-hiroshi, 2020-02-28:
Hi,

I found that when kinematic actor is moved by setKinematicTarget(), the event fired by entering or leaving a trigger is one frame later than when the move action happens.
If it is possible, I want the callback to be runned at the same frame but I do not know how to do this.

Any advice would be greatly appreciated.

kstorey-nvidia, 2020-02-28:
This is the expected behaviour. It ensures consistent behaviour with kinematic and dynamic rigid bodies. Set kinematic target instructs the simulation to move the kinematic to that pose in the next simulated frame. The actor doesn't immediatiately move to that location but, instead a velocity is calculated that will result in the actor reaching the target at the end of the next time-step. Collision detection (including trigger processing) is performed using the initial poses of the actors, the constraint solver is invoked and the rigid bodies are integrated by the velocities produced by the constraint solver.

You could get trigger events immediately by using setGlobalPose rather than setKinematicTarget but that would cause bad simulation behaviour when your kinematic interacted with dynamic rigid bodies (e.g. friction would not work properly because the kinematic would have teleported into a pose rather that moved continuously).

Just so we understand, what are the problems caused by not receiving trigger events until the next frame that you need to work around? If we understand the effect you are trying to achieve, we might be able to offer a solution.

sugano-hiroshi, 2020-03-03:
Thank you for your quick replay.

I understood the behaviour. Then, what I want to do is that when the kinematic moves then overlaps the trigger shape, I would like to get trigger events immediately after the simulation in the same frame. Trigger events fired in the next simulation is not tolerated for our game.

So, I guess it would be nice if there is a light functionality to detect and fire trigger events?

Levi-Armstrong, 2020-04-08:
@kstorey-nvidia I am running into this and I have tried the `setGlobalPose` but it does not trigger the simulation event callback `onContact`? How do you get the `setGlobalPose` to trigger the callback?