NVIDIAGameWorks/PhysX #258: Some internal wake counter reset values don't respect the user wakeCounterResetValue

thebusytypist, 2020-03-08:
When an actor is waken up, its wake counter is reset to a value, which can be controlled through `PxSceneDesc::wakeCounterResetValue`.
(https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxapi/files/classPxSceneDesc.html#a79e2c9c06f711272a48d7f07451117b7)

However I found internal implementation uses a fixed value (0.4 sec) no matter what I give to `wakeCounterResetValue`. Is it expected by design?

List of internal wake counter reset values:
```
PhysX/physx/include/PxSceneDesc.h
998:	wakeCounterResetValue				(20.0f*0.02f),

PhysX/physx/source/simulationcontroller/src/ScPhysics.cpp
40:const PxReal Sc::Physics::sWakeCounterOnCreation = 20.0f*0.02f;

PhysX/physx/source/simulationcontroller/src/ScBodySim.h
62:	static const PxReal ScInternalWakeCounterResetValue = 20.0f*0.02f;

PhysX/physx/source/lowleveldynamics/src/DyBodyCoreIntegrator.h
188:	PxReal wakeCounterResetTime = 20.0f*0.02f;
```

The one in `DyBodyCoreIntegrator.h` is especially important to me, as it controls the grace period before an actor goes to sleep. It would be better if I can make it longer by setting `PxSceneDescriptor::wakeCounterResetValue`.

tysoni, 2020-03-09:
Yes, I'd like to see an answer to this too. I raised the same issue (the hard-coded time counter) and other issues related to waking/sleeping here:

https://github.com/NVIDIAGameWorks/PhysX/issues/125

And never got a response...

msauter-nvidia, 2020-03-09:
This is mostly by design, yes. As documented, wakeCounterResetValue is only used for wakeUp() calls by the user. Internally, we would like to stay somewhat flexible in the way how dynamic objects get woken up and have a bit more tight control over how aggressive or relaxed the counters are set. Adaptive schemes to adjust the reset value per object might be an option, for example, as there are many factors that can be taken into account when internally waking up an object.
To tune sleeping behavior, I would recommend to try to use setSleepThreshold() and setStabilizationThreshold() first.
Last but not least, there are indeed some duplicates of the hardcoded internal reset value that we should unify.