NVIDIAGameWorks/PhysX #260: Linker error when building PhysX statically (PX_GENERATE_STATIC_LIBRARIES = True)

dantamont, 2020-03-09:
Hi folks, so I've just gotten sorted out with the proper run-time library settings between my library and PhysX (MDd and Md for debug and release, respectively), but am now running into some linker errors. I believe that I am linking against all required libraries, since this is everything that was generated by the build:

- PhysX_static_32
- PhysXCharacterKinematic_static_32
- PhysXCommon_static_32
- PhysXCooking_static_32
- PhysXExtensions_static_32
- PhysXFoundation_static_32
- PhysXPvdSDK_static_32
- PhysXVehicle_static_32

I am using Qt, and the relevant lines in my project settings are as follows:
```
# PhysX
INCLUDEPATH += ../../PhysX/physx/include \
			   ../../PhysX/pxshared/include
CONFIG(debug, debug|release) { 
	LIBS += -L$$PWD/lib/physx/debug -lPhysX_static_32
	LIBS += -L$$PWD/lib/physx/debug -lPhysXCharacterKinematic_static_32
	LIBS += -L$$PWD/lib/physx/debug -lPhysXCommon_static_32
	LIBS += -L$$PWD/lib/physx/debug -lPhysXCooking_static_32
	LIBS += -L$$PWD/lib/physx/debug -lPhysXExtensions_static_32
	LIBS += -L$$PWD/lib/physx/debug -lPhysXFoundation_static_32
	LIBS += -L$$PWD/lib/physx/debug -lPhysXPvdSDK_static_32
	LIBS += -L$$PWD/lib/physx/debug -lPhysXVehicle_static_32
}
```

I then call `#include <PxPhysicsAPI.h>` to summon the library.  However, I'm getting the following errors:

`Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK2019	unresolved external symbol __imp__PxCreateBasePhysics referenced in function "class physx::PxPhysics * __cdecl PxCreatePhysics(unsigned int,class physx::PxFoundation &,class physx::PxTolerancesScale const &,bool,class physx::PxPvd *)" (?PxCreatePhysics@@YAPAVPxPhysics@physx@@IAAVPxFoundation@2@ABVPxTolerancesScale@2@_NPAVPxPvd@2@@Z)	proj_name	C:\Users\me\Documents\Projects\proj_name\GbPhysicsManager.obj	1	`

`Error	LNK2019	unresolved external symbol __imp__PxRegisterArticulations referenced in function "class physx::PxPhysics * __cdecl PxCreatePhysics(unsigned int,class physx::PxFoundation &,class physx::PxTolerancesScale const &,bool,class physx::PxPvd *)" (?PxCreatePhysics@@YAPAVPxPhysics@physx@@IAAVPxFoundation@2@ABVPxTolerancesScale@2@_NPAVPxPvd@2@@Z)	proj_name	C:\Users\me\Documents\Projects\proj_name\GbPhysicsManager.obj	1	`

`Error	LNK2019	unresolved external symbol __imp__PxRegisterArticulationsReducedCoordinate referenced in function "class physx::PxPhysics * __cdecl PxCreatePhysics(unsigned int,class physx::PxFoundation &,class physx::PxTolerancesScale const &,bool,class physx::PxPvd *)" (?PxCreatePhysics@@YAPAVPxPhysics@physx@@IAAVPxFoundation@2@ABVPxTolerancesScale@2@_NPAVPxPvd@2@@Z)	proj_name	C:\Users\me\Documents\Projects\proj_name\GbPhysicsManager.obj	1	`

`Error	LNK2019	unresolved external symbol __imp__PxRegisterHeightFields referenced in function "class physx::PxPhysics * __cdecl PxCreatePhysics(unsigned int,class physx::PxFoundation &,class physx::PxTolerancesScale const &,bool,class physx::PxPvd *)" (?PxCreatePhysics@@YAPAVPxPhysics@physx@@IAAVPxFoundation@2@ABVPxTolerancesScale@2@_NPAVPxPvd@2@@Z)	proj_name	C:\Users\me\Documents\Projects\proj_name\GbPhysicsManager.obj	1	`

Any ideas?  I'm pretty stumped on this.  It's not a runtime library issue, and my library directory is showing up correctly in my project settings, which means that the linker should be searching in the correct place.

AlesBorovicka, 2020-03-09:
Hi,
 do you set preprocessor define PX_PHYSX_STATIC_LIB or include PxConfig.h before the physics headers?

dantamont, 2020-03-09:
Oh, thanks so much Ales!  I hadn't done that, I'm sure the documentation mentions that somewhere, but the build page didn't mention that the PX_PHYSX_STATIC_LIB was necessary.

closed 2020-03-09 by dantamont

SNMetamorph, 2023-09-13:
Thanks for solution, helps a lot!