NVIDIAGameWorks/PhysX #262: [Question] 2K Heightfield very slow

andre-ferreira, 2020-03-10:
Hello,

When trying to load an heightfield from a 2K heightmap the stepPhysics line gets unberably slow (it goes from 3ms (1K) in the logic update loop to 12ms (2K). The terrain size is 1024 (m). Is there anything i can do to improve performance or is this expected to happen?

Using PhysX 4.1.

kstorey-nvidia, 2020-03-10:
Please connect PVD (ideally one pass with all debug data and another with profile-only information) and share the PVD captures with us. This might help identify where the time is being spent. Depending on what you are doing, the number of samples may impact performance a little or a lot (e.g. if you have a bunch of bodies colliding with the environment and they just hit a few triangles each, resolution won't make much difference; if they hit 100s of triangles each, then it will impact performance significantly).

Also, can you confirm that you have not enabled debug visualization? This might get expensive if you have a very detailed heightfield because our debug viz code is not something that is optimized to be used all the time.

andre-ferreira, 2020-03-10:
> Please connect PVD (ideally one pass with all debug data and another with profile-only information) and share the PVD captures with us. This might help identify where the time is being spent. [...]

Hi!

First of all i must say i am very impressed with the response time of the PhysX team!
It was indeed the debug information that was killing the performance. As soon as i switched it off i got 1ms even with a 2K terrain.

Many thanks for your help!

closed 2020-03-10 by andre-ferreira