NVIDIAGameWorks/PhysX #263: TGS solver: constraining a dynamic rigidbody to a moving kinematic rigidbody will result in NaN velocity values on the dynamic body

tysoni, 2020-03-11:
Title explains the problem. This is happening in 4.1.1.

If you have a moving kinematic rigidbody and you constrain a dynamic rigidbody to it, the velocities on the rigidbody explode (quickly leading to inf/NAN values) after the constraint solver finishes. This only happens in the TGS solver. The PGS solver gives proper results.

tysoni, 2020-03-11:
Here are two videos showing the issue. The top box is a dynamic rigidbody. The bottom box is a kinematic rigidbody. The line between them shows the endpoints of the fixed constraint between them. I move the bottom box with calls to "setKinematicTarget". The only difference between the settings for these setups is the choice of solver.

http://www.tysonibele.com/Junk/PGS.mp4

http://www.tysonibele.com/Junk/TGS.mp4

kstorey-nvidia, 2020-03-11:
Thanks for reporting. We already reproduced and fixed this locally (this may even have come from a previous bug report on github). The fix has not yet been released on github.

In this method in DyTGSContactPrep.cpp:

PxU32 setupSolverConstraintStep(
	const PxTGSSolverConstraintPrepDesc& prepDesc,
	PxConstraintAllocator& allocator,
	const PxReal dt, const PxReal totalDt, const PxReal invdt, const PxReal invTotalDt,
	const PxReal lengthScale)

please change this line:

	s.velMultiplier += vel1;

to this

	s.velTarget += vel1;

Please confirm that this fixes your issue.

Thanks

tysoni, 2020-03-11:
That did the trick, thank you very much!

closed 2020-03-11 by tysoni