NVIDIAGameWorks/PhysX #264: Question: How best to leverage PhysX as a collision checker (No Physics)

Levi-Armstrong, 2020-03-11:
I am interested in leveraging PhysX for collision checking only, to be used with in robotic motion planning algorithms. I currently have implementations for FCL and Bullet but I am interested in the GPU support provided by PhysX. I am new to PhysX and I have been going through the API and testing out different functionalities. 

At a high level I would like to accomplish the following. Add RigidStatic objects to the environment along with RigidKinematic objects. Then I would like to move the RigidKinematics objects to different locations and at each location perform a collision/distance check. The information needed from the collision check is: distance/penetration depth, translation vector, contact point corresponding to the translation vector for each object in the contact pair.

I have been able to put a simple test where I add the RigidStatic objects and then use the scene overlap function to check contact with PxGeometry, but this only provide the depth and translation vector and not the contact points provided by GJK. I do see under geomutils that there are functions for getting this information but not sure what is the best approach this use case using PhysX. 

How should I approach this use case using PhysX?

shaneasd, 2020-03-16:
Have you considered using [split simulation](https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/Simulation.html#split-sim) to perform collision detection without resolution then use a [contact reporting](https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/AdvancedCollisionDetection.html#contact-reporting) callback to get the details of the collisions?

caveats:
- I haven't tried this so I'm not offering advice, just ideas
- the documentation linked is for physx 3 but you're presumably using physx 4. I'm assuming the interface for these features is fundamentally the same but haven't checked.
- The documentation for split simulation makes it seem like it would complain (or not work) if you keep detecting collisions without resolving them. You may need to keep recreating the scene to avoid this which may have significant performance implications.

Levi-Armstrong, 2020-03-16:
@shaneasd Thank you for the recommendation. I recently setup the split simulation, but I cannot get the onContact event to trigger. I copy the code into a single file so maybe so maybe you or someone else can see what I am doing incorrectly because the contact event it not  being calling. Now the contactReportFilterShader is being called so it is detecting the collision.

``` c++
#include <PxPhysicsAPI.h>
#include <istream>
#include <memory>

#define PX_RELEASE(x)	if(x)	{ x->release(); x = nullptr;}
#define UNUSED(x) (void)(x)
#define CONSOLE_BRIDGE_logError(x) { std::printf("%s\n", (x)); }

class TesseractSimulationEventCallback : public physx::PxSimulationEventCallback
{
public:
  void onConstraintBreak(physx::PxConstraintInfo* /*constraints*/, physx::PxU32 /*count*/) override {std::printf("onConstraintBreak\n");}
  void onWake(physx::PxActor** /*actors*/, physx::PxU32 /*count*/) override {std::printf("onWake\n");}
  void onSleep(physx::PxActor** /*actors*/, physx::PxU32 /*count*/) override {std::printf("onSleep\n");}
  void onTrigger(physx::PxTriggerPair* /*pairs*/, physx::PxU32 /*count*/) override {std::printf("onTrigger\n");}
  void onAdvance(const physx::PxRigidBody*const* /*bodyBuffer*/, const physx::PxTransform* /*poseBuffer*/, const physx::PxU32 /*count*/) override {std::printf("onAdvance\n");}

  // https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/AdvancedCollisionDetection.html#extracting-contact-information
  void onContact(const physx::PxContactPairHeader& pairHeader, const physx::PxContactPair* pairs, physx::PxU32 nbPairs) override
  {
    UNUSED(pairHeader);
    std::printf("TesseractSimulationEventCallback::onContact\n");
    const physx::PxU32 bufferSize = 64;
    physx::PxContactPairPoint contacts[bufferSize];
    for(physx::PxU32 i=0; i < nbPairs; i++)
    {
      const physx::PxContactPair& cp = pairs[i];
      physx::PxU32 nbContacts = pairs[i].extractContacts(contacts, bufferSize);
      for(physx::PxU32 j=0; j < nbContacts; j++)
      {
          physx::PxVec3 point = contacts[j].position;
          physx::PxVec3 impulse = contacts[j].impulse;
          physx::PxU32 internalFaceIndex0 = contacts[j].internalFaceIndex0;
          physx::PxU32 internalFaceIndex1 = contacts[j].internalFaceIndex1;
          UNUSED(point);
          UNUSED(impulse);
          UNUSED(internalFaceIndex0);
          UNUSED(internalFaceIndex1);
          //...
      }

      if(cp.events & physx::PxPairFlag::eNOTIFY_TOUCH_FOUND)
      {
  //            if((pairHeader.actors[0] == mSubmarineActor) ||
  //                (pairHeader.actors[1] == mSubmarineActor))
  //            {
  //                PxActor* otherActor = (mSubmarineActor == pairHeader.actors[0]) ?
  //                    pairHeader.actors[1] : pairHeader.actors[0];
  //                Seamine* mine =  reinterpret_cast<Seamine*>(otherActor->userData);
  //                // insert only once
  //                if(std::find(mMinesToExplode.begin(), mMinesToExplode.end(), mine) ==
  //                    mMinesToExplode.end())
  //                    mMinesToExplode.push_back(mine);

  //                break;
  //            }
      }
    }
  }
};

enum class FilterGroup : physx::PxU32
{
    eSTATIC     = (1 << 0),
    eKINEMATIC  = (1 << 1)
};

[... 188 lines cut ...]
                                                                     physx::PxReal(0.1),
                                                                     kin_shape_tf);
  kinematic_sphere->setName("kinematic_sphere");

  setupFiltering(phy, kinematic_sphere, static_cast<physx::PxU32>(FilterGroup::eKINEMATIC), static_cast<physx::PxU32>(FilterGroup::eSTATIC));
  phy.getScene()->addActor(*kinematic_sphere);

  for (int i = 0; i < 5; ++i)
  {
//    phy.getScene()->simulate(0.0001f);
    phy.getScene()->collide(0.0001f);
    phy.getScene()->fetchCollision(true);
    phy.getScene()->advance(); // Can this be skipped
    phy.getScene()->fetchResults(true);

    physx::PxTransform update_tf;
    update_tf.p = physx::PxVec3(0, 0, 1.0f - ((physx::PxReal(i) + 1.f) * 0.1f));
    update_tf.q = physx::PxQuat(0, 0, 1, 0);
    kinematic_sphere->setGlobalPose(update_tf);
  }
  std::printf("Done!\n");
  return 0;
}

```

kstorey-nvidia, 2020-03-16:
Does it work if you call setKinematicTarget rather than setGlobalPose()? PhysX only generates contacts between active actors, and setting globalPose (i.e. teleporting) a kinematic actor does not activate it.

Levi-Armstrong, 2020-03-16:
Great! That was the issue. One other question is it possible to omit the `advance()` call if the only object in the environment are static and kinematic?

kstorey-nvidia, 2020-03-16:
No, the advance call should do very minimal work, but it is still required even if you just have kinematic actors.

Levi-Armstrong, 2020-03-16:
Thank you.

So it runs now but after initial contact it does not call the onContact event if it remains in contact. Is it possible to set it up to always report contact even if it was previously in contact?

kstorey-nvidia, 2020-03-16:
This is because you asked it to do that with this flag:

Levi-Armstrong, 2020-03-16:
That was it. Thank you for you help. I will keep testing and integrating it open source software package.

shaneasd, 2020-03-16:
If you're forced to call advance anyway, I wonder if it would be simpler to forget about the split simulation and just use regular stepping. Presumably you get the same contact notifications at the same point in the physics logic.

Levi-Armstrong, 2020-03-16:
I will test just using the simulation method.

Levi-Armstrong, 2020-03-25:
The simulation seems to works as well.

Levi-Armstrong, 2020-03-27:
@kstorey-nvidia I have noticed if I add an `Actor` then call its `setKinematicTarget(new_tf)` followed by the call `simulate(0.0001f)` it reports the contact for the location of the actor prior to `setKinematicTarget`. Is there something I should call prior to simulate or do I just need to call simulate twice?

Levi-Armstrong, 2020-03-27:
Base on this [documentation](https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/Simulation.html?highlight=setkinematictarget) should you always call `wakeUp()` and `setWakeCounter()` after you call `setKinematicTarget()`?

shaneasd, 2020-04-11:
If you mean this bit https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxguide/Manual/Simulation.html?highlight=setkinematictarget#split-sim , I think that's just listing a bunch of write-through properties in no particular order. It's not intended to be a sequence of calls you should make. From my understanding I don't think it makes sense to call wake up on an actor you're moving kinematically. I believe wakeup is for dynamic objects whose dynamic objects are no longer being simulated because their energy and applied forces became negligibly low. This doesn't apply to kinematically moving objects because they essentially don't have dynamics, their energy is undefined because they have infinite inertia and they don't respond to forces.

Levi-Armstrong, 2020-04-15:
@shaneasd Thank you for the information. I wanting to use `setGlobalPose()` instead of `setKinemticPose()`, but when using `setGlobalPose()` it does not trigger a simulation onContact event. Do you know if it is possible to trigger the simulation `onContact` event when using setGlobalPose()?

shaneasd, 2020-04-16:
No I don't know.

However I'd be careful using setGlobalPose for this purpose. My understanding is setGlobalPose just teleports the object to the new transform. It has no record of where it's come from so no velocity. This can make collisions act in an unintuitive way because usually you want collisions to conserve momentum. This is probably why onContact isn't triggered because something like the contact velocity isn't obviously defined. setKinematicPose on the other hand basically says that over the course of a tick the object will have the required velocity to end up at the target pose so collisions are a bit easier to understand (but it does mean you don't want to use it to teleport things).

Given your context where you're only wanting collision checking (not resolution) I'm not sure which of the properties of the two methods are important but I would guess that at setKinematicPose is t least acceptable for a robot part moving what is presumably a small distance in a tick. Is there a good reason not to use it.

Levi-Armstrong, 2020-04-16:
The particular case I am using this for is a collision checker for a robotic motion planner. In this case the contact forces and velocities are not importance, but the contact points, distance or penetration at the end state are what I am after. Right now I am using the setKinematicPose which requires two simulation calls to get the contact information at the end location. I was hopping by using the setGlobalPose I could avoid making two calls to the simulate.

kstorey-nvidia, 2020-04-16:
You could potentially call setGlobalPose and setKinematicTarget. SetGlobalPose will move the actor to the location (so collision is performed in that pose) and setKinematicTarget will wake the kinematic actor up so that it actually generates contacts.

Levi-Armstrong, 2020-04-16:
I will tray that and see if it works.

shaneasd, 2020-04-17:
Presumably you could call wakeUp instead of setKinematicTarget if the issue is just that it's sleeping?

kstorey-nvidia, 2020-04-17:
Kinematic actors are a special-case in which sleeping state is based on whether they had a target set. It is illegal to explicitly call wakeUp() on a kinematic actor and will produce an error.

Levi-Armstrong, 2020-04-17:
@kstorey-nvidia your recommendation worked where I call `setGlobalPose()` followed by `setKinematicPose()` with the same transform eliminates the need to call simulation twice. Thank you.

Levi-Armstrong, 2020-04-20:
@kstorey-nvidia I have one more challenge to work through related to Octomaps. In our applications we leverage 3D perceptions for generating collision objects within the environment. This is usually represented by translating the octomap into several thousand boxed in the collision scene. 

I originally tried to add all of the boxes to a single actor but had segfault around 6,000 and not sure if there is a limit on subshape for actors. 

Then I switched to creating a single actor for each box and added it to an aggregate which seems to work but not sure what the best approach is. 

Any recommendations on how best to represent 3D perception data within PhysX?

kstorey-nvidia, 2020-04-20:
There is a 16-bit index involved in CmPtrTable, so the limit should be at 65,536 shapes per actor.

Do you have a callstack for the segfault with ~6000 shapes? Are you certain you didn't exceed 65,536?

I don't have any profile evidence to back-up what would be the best solution for this. However, my expectation would be that you can save some memory by using multiple shapes per actor. 

Having said that, I would not recommend bundling 1000s of shapes under a single actor. It's not a use-case we've actively optimized for.

Perhaps using some middle-ground of, say, ~128 shapes per actor combined with aggregates might be a good way to go.

Levi-Armstrong, 2020-04-20:
>Do you have a callstack for the segfault with ~6000 shapes? Are you certain you didn't exceed 65,536?

Sorry, typo it was around the 65,536. I have an octomap with around 400,000 occupied cells.

> Perhaps using some middle-ground of, say, ~128 shapes per actor combined with aggregates might be a good way to go.

I just want to make sure I understand correctly before implementing, so my planned approach is to break up the octomap into a set of actors with around 128 shapes. Then add all of the actors to a single aggregate, which ~400,000 shapes this would come to ~3125 actors. I have a few question below related to this approach.

* Do you see any issues with adding this many actors to a single aggregate? 

* When adding the actors with 128 shapes to the aggregate, do you recommend provided a BVH structure?

peiyongxian, 2020-12-21:
@Levi-Armstrong I am new to Physx, use physx as a collsion checker like you do, but i can not find the way to get distance/depth information in scene's overlap method. I would like to know how to get those informations? Can overlap method do? Or the only way is to call simulate() and processing contacts in callback function?

kstorey-nvidia, 2020-12-21:
The overlap method is just a yes/no overlap test. It does not provide penetration information.

You have a few options available that do not require calling simulate. You can either compute the MTD between a pair of shapes using PxGeometryQuery::computePenetration or you can use immediate mode to generate contacts using PxGenerateContacts.

Hope this helps

Levi-Armstrong, 2020-12-23:
I am currently using the simulate to process the contacts but I have used `PxGeometryQuery::computePenetration` in testing and plan to test the immediate mode. The reason I have stayed with using the simulate approach is I believe there are performance benefits that @kstorey-nvidia maybe can provided more details. If I understand correctly at the time simulate approach provided access to GPU acceleration and the other two approaches only ran on the CPU, but I am not sure this is true or not. @kstorey-nvidia are there any trade offs with performance between the three approaches?

peiyongxian, 2020-12-24:
Thank you for your answer.  I found that if there are not many geometries in  the environment (less than 100 in my test), collision checking using simulate() method with callback to check contacts on cpu is faster than that on gpu.

peiyongxian, 2020-12-24:
@kstorey-nvidia Really helps, thank you. I guess simulate() would be the better choice in the consideration of the broad phase algorithmn it includes which PxGenerateContacts does not have. Is that right?