NVIDIAGameWorks/PhysX #266: How to install PhysX 4.1 in Visual Studio 2019?

ngminteck, 2020-03-13:
Hi, i trying to install physX 4.1 in my school project.

i read the "platform_readme.html" file and check with preset folder, it don't have for Visual Studio 2019 version.

i do some search and found a person trying install in visual studio 2017 version 
https://stackoverflow.com/questions/58189023/physx-linking-problem-with-come-functions-imp-pxcreatebasephysics-reference

i follow and edit the XML file to V19 with these setting
`<?xml version="1.0" encoding="utf-8"?>
<preset name="vc19win64" comment="VC19 Win64 PhysX general settings">
  <platform targetPlatform="win64" compiler="vc19" />
  <CMakeSwitches>
    <cmakeSwitch name="PX_BUILDSNIPPETS" value="True" comment="Generate the snippets" />
    <cmakeSwitch name="PX_BUILDPUBLICSAMPLES" value="True" comment="Generate the samples projects" />
    <cmakeSwitch name="PX_GENERATE_STATIC_LIBRARIES" value="False" comment="Generate static libraries" />
    <cmakeSwitch name="NV_USE_STATIC_WINCRT" value="True" comment="Use the statically linked windows CRT" />
    <cmakeSwitch name="NV_USE_DEBUG_WINCRT" value="True" comment="Use the debug version of the CRT" />
    <cmakeSwitch name="PX_FLOAT_POINT_PRECISE_MATH" value="False" comment="Float point precise math" />
  </CMakeSwitches>
  <CMakeParams>
    <cmakeParam name="CMAKE_INSTALL_PREFIX" value="install/vc15win64/PhysX" comment="Install path relative to PhysX SDK root" />
  </CMakeParams>
</preset>`

After that i did not what to do.

I have Visual Studio 2019 16.4.29728.190, CMake 3.17 and Microsoft DirectX SDK June 2010  installed

i tried excute from CMake GUI application, but it failed.
https://drive.google.com/file/d/1U6mYJ_HatdX-GlR92oSd1LtBB_M6_ZG8/view?usp=sharing

AlesBorovicka, 2020-03-13:
Hi,
 Visual studio 2019 (vc16 (https://en.wikipedia.org/wiki/Microsoft_Visual_Studio)) solution can be generated using the vc16win64 preset.
Regards,
Ales

ngminteck, 2020-03-13:
Hi

should i just run  run generate_projects.bat?, i say it just says exit /b 0. as similar case with the forum link below
https://devtalk.nvidia.com/default/topic/1072498/physics-modeling/can-t-generatre-4-1-projects/

https://drive.google.com/file/d/1uLqisO8_FhXDjbTLLASZXcev4ombkcBH/view?usp=sharing

AlesBorovicka, 2020-03-13:
Probably this issue is related:
https://github.com/NVIDIAGameWorks/PhysX/issues/252
Maybe you need to try the workaround mentioned there

ngminteck, 2020-03-13:
Thank, i manage to run the .bat

closed 2020-03-13 by ngminteck

xcplay, 2020-08-14:
add vc16win64.xml in buildtools\presets\public dictionary
`<?xml version="1.0" encoding="utf-8"?>
<preset name="vc16win64" comment="VC16 Win64 PhysX general settings">
  <platform targetPlatform="win64" compiler="vc16" />
  <CMakeSwitches>
    <cmakeSwitch name="PX_BUILDSNIPPETS" value="True" comment="Generate the snippets" />
    <cmakeSwitch name="PX_BUILDPUBLICSAMPLES" value="True" comment="Generate the samples projects" />
    <cmakeSwitch name="PX_GENERATE_STATIC_LIBRARIES" value="False" comment="Generate static libraries" />
    <cmakeSwitch name="NV_USE_STATIC_WINCRT" value="True" comment="Use the statically linked windows CRT" />
    <cmakeSwitch name="NV_USE_DEBUG_WINCRT" value="True" comment="Use the debug version of the CRT" />
    <cmakeSwitch name="PX_FLOAT_POINT_PRECISE_MATH" value="False" comment="Float point precise math" />
  </CMakeSwitches>
  <CMakeParams>
    <cmakeParam name="CMAKE_INSTALL_PREFIX" value="install/vc16win64/PhysX" comment="Install path relative to PhysX SDK root" />
  </CMakeParams>
</preset>`

add your vs2019 compiler in cmake_generate_projects.py
`def getPlatformCMakeParams(self):
        outString = ' '
        if self.compiler == 'vc12':
            outString = outString + '-G \"Visual Studio 12 2013\"'
        elif self.compiler == 'vc14':
            outString = outString + '-G \"Visual Studio 14 2015\"'
        elif self.compiler == 'vc15':
            outString = outString + '-G \"Visual Studio 15 2017\"'
        elif self.compiler == 'vc16':
            outString = outString + '-G \"Visual Studio 16 2019\"'`

cibolar, 2021-05-31:
Hi friends. I have the same problem, I want to instal PhysX in Visual Studio 2019. First I installed CMAKE, and after that I run the generate_projects.bat file in PhysX, and I selected the vc16win64 preset. What else should I do to be make it ready to use in Visua Studio? Because after this steps, I open a VS file, and I include #include <PxPhysicsAPI.h>, but it gives error.