NVIDIAGameWorks/PhysX #268: No collisions between two meshes

mirkow, 2020-03-17:
Hi,

we have the problem that two objects that consists of meshes instead of primitives do not collide with each other.
Collisions between meshes and primitives work as well as collision between primitives.

Is this expected behaviour? And if not what might be the reason that it does not work?

Cheers

Levi-Armstrong, 2020-03-27:
Not sure if this is relavent but the documentation makes the two statements below.

    

- TriangleMesh, HeightField and Plane geometries are not supported for simulation shapes that are attached to dynamic actors, unless the dynamic actors are configured to be kinematic.

- TriangleMesh and HeightField geometries are not supported for trigger shapes.

mirkow, 2020-03-29:
Yes, with cooking of convex meshes instead of triangles it works. But what does this doc comment mean that it can only have 256 vertices? That is basically nothing for usual meshes. What happens if a mesh has more vertices?

Levi-Armstrong, 2020-03-30:
I believe this limit is for it to run on the GPU. If it is larger it runs on the CPU. I am not sure if this limit is configurable.

kstorey-nvidia, 2020-03-31:
Arbitrary triangle meshes have no limits on the number of vertices or triangles. However, these can only be used as collision geometries for static actors. Dynamic actors can only be constructed from a set of convex primitives (sphere, capsule, box and convex mesh). Convex meshes have a hard limit of 256 vertices. PhysX provides functionality to generate a convex mesh from an arbitrary input mesh. If a single convex will not approximate your shape sufficiently, you can use a library like VHACD to produce a convex decomposition and you can then provide a set of convex shapes for an actor to use as its collision.

It is also worth noting that PhysX does support using different shapes for scene query and simulation, and it is legal to use an arbitrary triangle mesh as the scene query shape for a dynamic body. This means that you can perform queries like raycasts or sweeps against a full-detail triangle mesh while using a simplified convex - or convex decomposition - for the simulation shapes.

mirkow, 2020-04-01:
> Convex meshes have a hard limit of 256 vertices. PhysX provides functionality to generate a convex mesh from an arbitrary input mesh.

Thanks, which function would that be?

internetfx, 2020-07-20:
First come up with an array of vertices. You should have a PxCooking object which will "cook" it. Make a PxConvexMeshDesc descriptor so that the cooking routing knows where the vertices are, along with the length and stride. Create a PxDefaultMemoryOutputStream and call pxCooking->cookConvexMesh(...) to cook it. Copy the output stream to a PxDefaultMemoryInputData and have pxPhysics->createConvexMesh(...) on that input to make a PxConvexMesh. You can now construct a PxConvexMeshGeometry on the convex mesh. Finally, you can pxPhysics->createShape(...) using that mesh geometry.

A code snippet I'm using from an experiment:

```
PxShape *Physics::CreateConvexShape(const PxVec3 *vertices, PxU32 size, const PxMaterial &pxMaterial)
{
	// build descriptor for how to read vertices

	PxConvexMeshDesc convexMeshDesc;
	convexMeshDesc.points.count = size;
	convexMeshDesc.points.stride = sizeof(PxVec3);
	convexMeshDesc.points.data = vertices;
	convexMeshDesc.flags = PxConvexFlag::eCOMPUTE_CONVEX;

	// cook mesh to output stream

	PxDefaultMemoryOutputStream cookedMeshOutput;
	PxConvexMeshCookingResult::Enum cookingResult;
	if (!mCooking->cookConvexMesh(convexMeshDesc, cookedMeshOutput, &cookingResult))
		throw runtime_error("Could not cook convex mesh");

	// use output as input to convex mesh

	PxDefaultMemoryInputData cookedMeshInput(cookedMeshOutput.getData(), cookedMeshOutput.getSize());
	PxConvexMesh *convexMesh = mPhysics->createConvexMesh(cookedMeshInput);

	// create shape from conves mesh geometry

	PxShape *convexMeshShape = mPhysics->createShape(PxConvexMeshGeometry(convexMesh), pxMaterial,
		false /* not exclusive */);

	// clean up

	convexMesh->release();

	// return shape
	
	return convexMeshShape;
}
```