NVIDIAGameWorks/PhysX #275: How to get contact point for both shapes from PxContactPairPoint?

Levi-Armstrong, 2020-03-27:
Currently PxContactPairPoint provides the position in world but it does not indicate what shape it is on. Usually with GJK algorithm you get a contact point on both shapes. I noticed that the normal says it points from the second shape to the first shape so assuming that the position is always provided for the second shape you could calculate the point for the first shape. I have tested this approach but it does not appear that the contact point is consistently provided for the second shape.

Tom-Forsyth, 2022-09-27:
Were you ever able to figure this one out? I know this was quite a bit ago, but I am having this same issue and did not want to open a new issue.

I have checked out #264 , and I have essentially the same use case. I am implementing collision avoidance in motion planning, but I am having trouble figuring out if a contact point is with respect to the robot or the obstacle in my callback. All I am able to identify is if the normal points away from the robot or not based on the order of the shapes like you mentioned in the post.

For now I am just using a very large minimum allowed distance in my planner, but I am needing more precision.

I do not need both contacts, but rather just the knowledge of which shape the contact belongs to so that I can translate the contact point along the normal by the separation to calculate the second contact.

Tom-Forsyth, 2022-10-03:
I see, I will do some poking around and see if I can figure this out.