NVIDIAGameWorks/PhysX #279: Error when running physx/generate_projects.sh

Bicyclette, 2020-04-01:
Hi, I tried to run the **generate_projects.sh** script, and I had the following error:

`
Using preset xml: buildtools/presets/public/linux.xml
Target platform: linux using compiler: clang
PM_CMakeModules_PATH: /home/mathias/Programmation/PhysX/physx/../externals/cmakemodules
PM_PATHS: /home/mathias/Programmation/PhysX/physx/../externals/opengl-linux;/home/mathias/Programmation/PhysX/physx/../externals/targa;/home/mathias/Programmation/PhysX/physx/../externals/cg-linux;/home/mathias/Programmation/PhysX/physx/../externals/glew-linux
Cmake: cmake
Not searching for unused variables given on the command line.
-- The C compiler identification is Clang 9.0.1
-- The CXX compiler identification is Clang 9.0.1
-- Check for working C compiler: /usr/bin/clang
-- Check for working C compiler: /usr/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++
-- Check for working CXX compiler: /usr/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
PHYSX ROOT /home/mathias/Programmation/PhysX/physx
PhysX Build Platform: linux
Using CXX Compiler: /usr/bin/clang++
CMake Error at /home/mathias/Programmation/PhysX/externals/cmakemodules/GetCompilerAndPlatform.cmake:120 (IF):
  if given arguments:

    "STREQUAL" "x86_64-unknown-linux-gnu" "OR" "STREQUAL" "x86_64-linux-gnu"

  Unknown arguments specified
Call Stack (most recent call first):
  /home/mathias/Programmation/PhysX/externals/cmakemodules/NvidiaBuildOptions.cmake:99 (GetPlatformBinName)
  /home/mathias/Programmation/PhysX/physx/source/compiler/cmake/CMakeLists.txt:70 (INCLUDE)

-- Configuring incomplete, errors occurred!
See also "/home/mathias/Programmation/PhysX/physx/compiler/linux-debug/CMakeFiles/CMakeOutput.log".
`

How can I fix that ? I tried doing what is discussed on that link => [https://github.com/NVIDIAGameWorks/PhysX/issues/39](url) without any success (I'm using Fedora 31).

Thank you in advance !

Bicyclette, 2020-04-02:
When I replace ${CMAKE_LIBRARY_ARCHITECTURE} by ${CMAKE_CXX_COMPILER}
I get the following message
```
Not searching for unused variables given on the command line.
-- The C compiler identification is Clang 9.0.1
-- The CXX compiler identification is Clang 9.0.1
-- Check for working C compiler: /usr/bin/clang
-- Check for working C compiler: /usr/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/clang++
-- Check for working CXX compiler: /usr/bin/clang++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
PHYSX ROOT /home/mathias/Programmation/PhysX/physx
PhysX Build Platform: linux
Using CXX Compiler: /usr/bin/clang++
CMake Warning (dev) at /home/mathias/Programmation/PhysX/physx/source/compiler/cmake/linux/CMakeLists.txt:36 (SET):
  implicitly converting 'INTERAL' to 'STRING' type.
Call Stack (most recent call first):
  /home/mathias/Programmation/PhysX/physx/source/compiler/cmake/CMakeLists.txt:130 (INCLUDE)
This warning is for project developers.  Use -Wno-dev to suppress it.

Added PhysX
CMake Error at /home/mathias/Programmation/PhysX/physx/snippets/compiler/cmake/linux/CMakeLists.txt:56 (IF):
  if given arguments:

    "STREQUAL" "aarch64-linux-gnu"

  Unknown arguments specified
Call Stack (most recent call first):
  /home/mathias/Programmation/PhysX/physx/snippets/compiler/cmake/CMakeLists.txt:80 (INCLUDE)

-- Configuring incomplete, errors occurred!
See also "/home/mathias/Programmation/PhysX/physx/compiler/linux-release/CMakeFiles/CMakeOutput.log".
```

khadoran, 2021-01-15:
I ran into the same problem. There are two lines in the mentioned CMakeLists.txt file starting with SET(PHYSX_CXX_FLAGS ...) where the word internal is misspelled -> CACHE INTERAL (which should be CACHE INTERNAL)

Fixing this in-place silenced these warnings for me