NVIDIAGameWorks/PhysX #280: How do I use triangular meshes on a dynamic object?

mrjbom, 2020-04-03:
Hi.
Can't I use triangular meshes on a dynamic body? I want a body whose shape is described by a triangular mesh to be able to work as dynamic, not as kinematic or static, so that it flies away from impacts and obeys the force of gravity.

Bicyclette, 2020-04-03:
Hi !
First of all, I've never used Physx, I'm currently learning Bullet, but maybe it works the same way, so I thought I could share my thoughts about that, maybe it will help.
I asked myself the same question a few weeks ago, and I eventually found out that I had to create a group of convex subshapes, resulting in a concave dynamic object as a whole.
In Bullet you create a btCompoundShape, and then you add some child shapes to it (all those child shapes are convex).
Maybe you should do the same thing in Physx ? I don't know, I hope someone will help you with that.

mrjbom, 2020-04-04:
I've thought of that, too. I wanted to make a set of triangles and connect them into a single shape. Thanks for the answer.

Bicyclette, 2020-04-04:
You're welcome ;)

amoravanszky, 2020-04-17:
I confirm that presently the Bicyclette's approach is your only option.  You can use a third party library like VHACD to generate the decomposition.

mrjbom, 2020-04-18:
I tried to use V-HACD for this purpose, but I encountered that the convex parts generated by this library behave incorrectly. Details of this problem are described here: https://github.com/NVIDIAGameWorks/PhysX/issues/287

mrjbom, 2020-04-19:
Thank you so much for your help.

closed 2020-04-19 by mrjbom