NVIDIAGameWorks/PhysX #281: Massive spikes in postBroadPhaseStage3

LightStriker, 2020-04-10:
I'm running out of option, as Unity appears to not know what's going on.

We have game in Unity - which I believe is running PhysX 4.0 or 4.1? - and we have the following happening in 2 scenes; https://i.imgur.com/TzVwqQK.png
The game is a third-person RPG. Think "Dark Souls"-like type of game, where we swing swords around.

We have massive spikes happening in "postBroadPhaseStage3" and we have no idea what's causing them. What we managed to figured out from out tests:
- Something in the scene breaks the broadphase as not all scene have the issue
- The spikes happens when triggers overlaps each others, like when two characters attack each others
- The number of character or number of overlapping triggers only make those spikes appear more often.
- The size of those spike is constant to that scene. Removing specific sector of the scene reduce the size of those spike. 

What exactly happens in postBroadPhaseStage3? And what's in our scene that could make it go nuts?

kstorey-nvidia, 2020-04-10:
Are you using ccd? That code doesn't look like it should do much apart from during a ccd pass. If you are using ccd, can you try disabling it temporarily to confirm whether this is the case? Does this spike disappear, or just move somewhere else?

My guess would be that a very large number of new pairs are being generated or lost in that frame.

This could be a byproduct of ccd massively expanding and reducing the bounds of some shapes, but there isn't enough information from your description to make anything more than a guess.

PhysX has a tool called pvd that we use to capture games that provides the ability to scrub through frames to identify the source of performance issues. This would be ideal for this case because you could attach a capture to this issue and it might help answer your problem very quickly by understanding what exactly happened on the spiking frame - a big allocation, a sudden spikes in the number of pairs etc.

Unfortunately, as far as I know, unity doesn't support pvd so that isn't an option.

LightStriker, 2020-04-10:
CCD as in continuous collision detection? I believe so, the capsule of our characters are in that mode.

However, those spikes happens when triggers overlaps, even if they aren't flagged as to interact with each other. Actually, nothing between those characters are supposed to interact.

Our hit detection in an attack is performed using boxcast against the triggers layer. However, the issue happens even if we disable the boxcast.

LightStriker, 2020-04-10:
We removed all enemies... and we managed to reproduce the spikes with only the player running around. A spike happens anytime one of its hitbox trigger intersect with a collider in that specific map.

What's really annoying is, those hitboxes are on a layer where they are not supposed to interact with anything. Is it possible to do boxcast against disabled collider? (stupid idea like that)

kstorey-nvidia, 2020-04-10:
I'm afraid I don't have any gut feeling as to what is making this take so long. If you are using continuous collision detection, please turn it off as an experiment. That code processes all the user events for lost touches when a pair of bounds stops to overlap. When CCD is disabled, it really doesn't do much. That would at least give us some indication of where the time is going.

Also, can you please confirm that there are not any error messages being emitted from PhysX? Spikes could be caused by error logging. It is possible PhysX is emitting an error message advising you that there is some illegal operation being performed (e.g. trying to request the solver to solve a collision between 2 infinite-mass bodies).

>>What's really annoying is, those hitboxes are on a layer where they are not supposed to interact with anything. Is it possible to do boxcast against disabled collider? (stupid idea like that)

It is possible to have shapes that only appear in scene queries with PhysX. Shapes can be assigned to simulation, scene query or both using PxShapeFlag::eSIMULATION_SHAPE and PxShapeFlag::eSCENE_QUERY_SHAPE. If the shapes appear in scene query only, they will not be present in the simulation scene and no overlaps will be detected. However, I am not very familiar with unity so I am not able to advise how you could leverage this functionality within unity.

LightStriker, 2020-04-10:
Yeah, sadly Unity didn't expose anything about shapes. It all goes through colliders.

Also tried turning off CCD; no luck.

As for illegal operation, if it's logged it's now logged by Unity.

I don't understand why triggers intersecting with colliders would do anything, especially since they are not in any interacting layers.