NVIDIAGameWorks/PhysX #282: Development of Tire Contact Model on Soft Soil

utbeaver, 2020-04-13:
Hi, we are working on the support of soft soil modeling in PhysX and function we are looking at is PxVehicleComputeTireForce (https://docs.nvidia.com/gameworks/content/gameworkslibrary/physx/guide/Manual/Vehicles.html#tire-shaders).

I have looked into teh example of the implementation. However, in order to models oft soil, we need to know the terrain type and tire perpetration into the ground. How could we get them?

Thanks for attention,

amoravanszky, 2020-04-17:
Did you see we have this tutorial on making soft soil: 

https://developer.nvidia.com/content/physx-dynamic-heightfield-modifications

utbeaver, 2020-04-18:
Thanks!  It looks like what I am looking at.  I will take a dive into and come back.

Adam,  my hat off to you, a respected collegue. We might meet some day in this small world.

utbeaver, 2024-11-11:
It seems code link is broken.

ceccocats, 2025-04-17:
@utbeaver 
Do you get to find the source code of that demo?