NVIDIAGameWorks/PhysX #287: PhysX bug? Spontaneous rotation of the convex hull.

mrjbom, 2020-04-14:
Hi.
I'm not sure if this is a PhysX problem, but it looks a lot like it.
I'll start from afar.
To use non-convex objects in my program, I split the object into convex parts using V-HACD(triangular grids do not allow modeling these objects as dynamic).
Here is an example of a shape split using V-HACD.
[image]

For the test, I take the left part of this figure, this is how the data about its vertexes looks in the .wrl file.
```
ccw TRUE
solid TRUE
convex TRUE
coord DEF co Coordinate {
          point [
                        -0.000000 1.295039 -5.117647,
                        -9.881939 10.235294 5.235294,
                        -10.117647 10.235294 5.235294,
                        -10.117647 -0.117647 5.235294,
                        -10.117647 10.235294 -5.117647,
                        -10.117647 -0.117647 -5.117647,
                        -0.000000 -0.116633 5.235294,
                        -0.000000 -0.116633 -5.117647,
                        -0.000000 1.295039 5.235294,
                        -6.353240 7.175826 -5.117647,
                        -1.646990 2.941821 5.235294,
                        -1.646990 2.941821 -5.117647,
                        -9.881939 10.235294 -5.117647,
                        -8.469667 9.057718 5.235294,
                        -8.469667 9.057718 -5.117647,
                        -6.353240 7.175826 5.235294,
                    ]
                }
```
I have manually rewritten them and am trying to use them to create the first half of the split shape.
It is successfully created and looks correct, but immediately turns over absurdly under the influence of unknown forces.

Here is the full source code.
```
#include <iostream>
#include <fstream>
#include <vector>
#include <PxPhysicsAPI.h>
#include "ASEFile.h"

using namespace physx;
using namespace std;

#define RadiansToDegress(radians) radians * 180.0 / 3.141592
#define DegressToRadians(degrees) degrees * 3.141592 / 180.0
PxQuat ToQuaternion(float, float, float);

PxDefaultErrorCallback	gDefaultErrorCallback;
PxDefaultAllocator		gDefaultAllocatorCallback;

PxFoundation* gFoundation = NULL;
PxPhysics* gPhysics = NULL;
PxCooking* gCooking = NULL;

PxDefaultCpuDispatcher* gDispatcher = NULL;

PxPvd* gPvd = NULL;
PxPvdTransport* gPvdTransport = NULL;
PxPvdSceneClient* gPvdClient = NULL;

int main()
{
    //create PxFoundation
    gFoundation = PxCreateFoundation(PX_PHYSICS_VERSION, gDefaultAllocatorCallback,
        gDefaultErrorCallback);
    if (!gFoundation) {
        cout << "PxCreateFoundation failed!\n";
        return -1;
    }
    else {
        cout << "PxCreateFoundation success\n";
    }

    //create PxPvd
    bool recordMemoryAllocations = true;

    gPvd = PxCreatePvd(*gFoundation);
    PxPvdTransport* transport = PxDefaultPvdSocketTransportCreate("localhost", 5425, 10);
    gPvd->connect(*transport, PxPvdInstrumentationFlag::eALL);

    //create PxPhysics
    PxTolerancesScale scale;
    scale.length = 1.f;          // typical length of an object
    scale.speed = 9.81f;         // typical speed of an object, gravity*1s is a reasonable choice

    gPhysics = PxCreatePhysics(PX_PHYSICS_VERSION, *gFoundation,
        scale, recordMemoryAllocations, gPvd);
    if (!gPhysics) {
        cout << "PxCreatePhysics failed!\n";
        return -1;
    }
    else {
        cout << "PxCreatePhysics success\n";
    }

    //create PxCooking
    gCooking = PxCreateCooking(PX_PHYSICS_VERSION, *gFoundation, PxCookingParams(scale));
    if (!gCooking) {
        cout << "PxCreateCooking failed!\n";
        return -1;
    }
    else {
        cout << "PxCreateCooking success\n";
    }

    gDispatcher = PxDefaultCpuDispatcherCreate(2);

    PxMaterial* myMaterial = gPhysics->createMaterial(1.0, 1.0, 0.5);

    PxMaterial* groundMaterial = gPhysics->createMaterial(1.0, 1.0, 0.5);
    PxShape* groundShape = gPhysics->createShape(PxPlaneGeometry(), *myMaterial, true);

    PxSceneDesc sceneDesc(gPhysics->getTolerancesScale());
    sceneDesc.gravity = PxVec3(0.0f, -9.81f, 0.0f);
    gDispatcher = PxDefaultCpuDispatcherCreate(2);
    sceneDesc.cpuDispatcher = gDispatcher;
    sceneDesc.filterShader = PxDefaultSimulationFilterShader;
    PxScene* myScene = gPhysics->createScene(sceneDesc);

    gPvdClient = myScene->getScenePvdClient();
    if (gPvdClient)
    {
        gPvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_CONSTRAINTS, true);
        gPvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_CONTACTS, true);
        gPvdClient->setScenePvdFlag(PxPvdSceneFlag::eTRANSMIT_SCENEQUERIES, true);
    }

    PxRigidStatic* groundPlane = PxCreatePlane(*gPhysics, PxPlane(0, 1, 0, 0), *myMaterial);
    myScene->addActor(*groundPlane);

    static const PxVec3 convexVerts2[] = {
        PxVec3(-0.000000, 1.295039, -5.117647),
        PxVec3(-9.881939, 10.235294, 5.235294),
        PxVec3(-10.117647, 10.235294, 5.235294),
        PxVec3(-10.117647, -0.117647, 5.235294),
        PxVec3(-10.117647, 10.235294, -5.117647),
        PxVec3(-10.117647, -0.117647, -5.117647),
        PxVec3(-0.000000, -0.116633, 5.235294),
        PxVec3(-0.000000, -0.116633, -5.117647),
        PxVec3(-0.000000, 1.295039, 5.235294),
        PxVec3(-6.353240, 7.175826, -5.117647),
        PxVec3(-1.646990, 2.941821, 5.235294),
        PxVec3(-1.646990, 2.941821, -5.117647),
        PxVec3(-9.881939, 10.235294, -5.117647),
        PxVec3(-8.469667, 9.057718, 5.235294),
        PxVec3(-8.469667, 9.057718, -5.117647),
        PxVec3(-6.353240, 7.175826, 5.235294)
    };

    PxConvexMeshDesc convexDesc;
    convexDesc.points.count = sizeof(convexVerts2) / sizeof(PxVec3);
    convexDesc.points.stride = sizeof(PxVec3);
    convexDesc.points.data = convexVerts2;
    convexDesc.flags = PxConvexFlag::eCOMPUTE_CONVEX;

    PxDefaultMemoryOutputStream buf;
    PxConvexMeshCookingResult::Enum result;
    if (!gCooking->cookConvexMesh(convexDesc, buf, &result))
        return NULL;
    PxDefaultMemoryInputData input(buf.getData(), buf.getSize());
    PxConvexMesh* convexMesh = gPhysics->createConvexMesh(input);

    PxShape* myShape2 = gPhysics->createShape(PxConvexMeshGeometry(convexMesh), *myMaterial);
    PxRigidDynamic* myRigidDynamic2 = gPhysics->createRigidDynamic(PxTransform(PxVec3(0.f, 3.0f, 0.f)));
    myRigidDynamic2->attachShape(*myShape2);
    myScene->addActor(*myRigidDynamic2);

    for (unsigned int i = 0; i < 5000; ++i)
    {
        myScene->simulate(1.0f / 60.0f);
        myScene->fetchResults(true);
    }

    //shutdown
    gCooking->release();
    gPhysics->release();
    gPvd->release();
    gFoundation->release();
}
```
Visual Debugger screenshots:
[image]
[image]
[image]
Thanks.

kstorey-nvidia, 2020-04-18:
In your code snippet, you do not appear to be computing the center of mass, mass or inertia of the rigid body. If you don't compute this, the default mass/inertia values are unlikely to behave appropriately for any shape besides a simple spherical shape.

Make sure you call PxRigidBodyExt::updateMassAndInertia() on the rigid body and hopefully the motion should look more correct.

mrjbom, 2020-04-20:
Thanks, love you♥

closed 2020-04-20 by mrjbom