NVIDIAGameWorks/PhysX #289: Double sided collision of static triangular meshes

kkdb8dm, 2020-04-17:
I am working on version 3.4 and  trying to diagnose an issue in collisions being missed between static triangular meshes and dynamic objects. Here is the unit test I set up.

A static PxTrangleMesh constructed using a single triangle with planar points ([0,0, 0], [0, 0, 10], [10, 0, 10]). And then I drop a dynamic actor with a box type collision model from height. I expect the box to collide with this triangle and rest on it. But my observation is this.

* If the verts are specified in counter clock wise direction, the box does collide and comes to rest at exactly the location I expect.
* If I specify the verts in clock wise order, the box does not collide at all and passes right through this triangle.

My questions.
1. Should I then conclude that the default collision between static Triangular meshes and dynamic actors are one-sided with the triangle normals being determined by the order in which I specify the vetrs?
2. If answer to 1 is yes, how do I make the collision double sided so that the order in which I specify the verts does not matter?

Or is there some other obvious explanation which I have missed here or there could be bugs elsewhere in my code?

MrMechanical, 2020-05-26:
Out of curiosity what happens if you put 2 triangles facing each other? Does it then detect collisions double-sided?

kkdb8dm, 2020-06-24:
Yes. That works.