NVIDIAGameWorks/PhysX #297: VS2019 Compile error

tomeaton17, 2020-05-05:
I have been trying to get PhysX to compile on VS2019 for the last 2 days. I fixed the "typeinfo.h" issue but now I am having a problem with D3D11Renderer.cpp. The problem I am having is this 

`Warning	C5205 delete of an abstract class 'SampleRenderer::Any::Proxy' that has a non-virtual destructor results in undefined behavior (compiling source file C:\Users\TomEaton\Desktop\PhysX4.1\physx\samples\sampleframework\renderer\src\d3d11\D3D1RendererMesh.cpp)` on line 53. 

`-Wx` compiler flag is being used so this warning is being treated as an error. Is this a problem with the way I generated the visual studio project?

How should I continue?

PathogenDavid, 2020-05-11:
I ran into this too. Judging by the high warning number and the [lack of documentation](https://docs.microsoft.com/en-us/cpp/error-messages/compiler-warnings/compiler-warnings-c4800-through-c4999?view=vs-2019), I'm inclined to assume this warning (as well as the others in SampleRenderer) were introduced in an update to Visual Studio 2019.

To work around the issue, I just disabled warnings-as-errors for the sample projects by adding `/WX-` to [physx/samples/compiler/cmake/windows/CMakeLists.txt:49](https://github.com/NVIDIAGameWorks/PhysX/blob/ae80dede0546d652040ae6260a810e53e20a06fa/physx/samples/compiler/cmake/windows/CMakeLists.txt#L49) and regenerating the projects.

```cmake
SET(CMAKE_CXX_FLAGS "${PHYSX_CXX_FLAGS}  /WX-  /wd4245 /wd4512 /wd4996 /wd4244 /wd4100 /wd4456 /wd4189 /wd4505 /wd4389 /EHsc ${ADDL_WARNINGS}")
```

tomeaton17, 2020-05-18:
Thanks, David. I gave up on trying to compile PhysX myself and just used `vcpkg` instead.

closed 2020-05-18 by tomeaton17