NVIDIAGameWorks/PhysX #298: Why is PxPvdRenderBuffer.h considered an internal header?

cjtallman, 2020-05-08:
The PVD user renderer debug types (PvdDebugPoint, PvdDebugTriangle, etc) have forward declarations in [PxPvdSceneClient.h](https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxapi/files/classPxPvdSceneClient.html), but their definitions aren't actually included in the public SDK, so there's no way to use them.

I discovered PxPvdRenderBuffer.h is grouped in PHYSXPVDSDK_INTERNAL_HEADERS in the cmake file, and am wondering if that's a mistake or intentional.

amoravanszky, 2020-09-11:
I don't know of anyone besides our PVD team using our PVD API, it was really only intended to support our own PVD application.  That might be the reason why its a little rough.  The visualization users should implement in their engines is intended to be

https://gameworksdocs.nvidia.com/PhysX/4.1/documentation/physxapi/files/classPxRenderBuffer.html

cjtallman, 2020-09-21:
Thanks for the reply. We're using the debug rendering in PhysX, but I was trying to figure out how to draw debug primitives in PVD, as in cases where there's a headless simulation. 

I was confused because it mentions these API's in the PVD help docs though they didn't seem to be available to actually use.
Sorry I can't find a web link to it. See the screenshot from the PVD help screen:
[image]