NVIDIAGameWorks/PhysX #299: Convex shapes not coming to rest after migrating from 3.4 to 4.1

jankrassnigg, 2020-05-19:
We recently migrated from PhysX 3.4 to 4.1 and are now seeing some objects with convex meshes jittering endlessly in place.

This only seems to happen if the object gets a little bit of rotation, though. I recorded this behavior here:
[restless.zip](https://github.com/NVIDIAGameWorks/PhysX/files/4652725/restless.zip)

Any idea why this happens and what change in PhysX 4 made this happen? I know that convex meshes now don't have a skin width anymore, and I tried different restOffsets and contactOffsets, but nothing had a positive effect. Do we need to adjust the convex hull generation maybe?

kstorey-nvidia, 2020-05-21:
Thanks for reporting.

The convex mesh collision semantics should be identical between PhysX 3.4 and 4.1. The semantics changed slightly between 3.3 and 3.4.

From looking at your capture, the misbehaving convex seems to be at the boundary between two triangles (you can see this in PVD if you enable wireframe rendering). I am not aware of any new reported instabilities with convex vs mesh in PhysX 4 but I will file a bug with your PVD capture as a repro step and hopefully it will be looked into soon.

One observation I can make is that the triangles on your PxTriangleMesh are quite large. Could you try a more tessellated triangle mesh to see if this improves things? Very large ratios between the polygons involved could cause some numerical instabilities, but AFAIK, not a lot changed in this code between 3.4 and 4.1.

jankrassnigg, 2020-05-21:
Hey @kstorey-nvidia thanks for the reply. I tried a couple of things to narrow this down. See the attached captures:
[more restless tests.zip](https://github.com/NVIDIAGameWorks/PhysX/files/4661698/more.restless.tests.zip)

Dropping the convex onto box shapes, both static and dynamic, works fine and they come to rest quickly. For the triangle mesh I have reduced the triangle size. I would expect this size to be fine?

As you can see, some of the restless objects are on triangle edges, but some are not. It does seem that all that are on an edge end up restless, though.

Hope that helps.

kstorey-nvidia, 2020-05-21:
Thanks for the extra details. I've tracked this in a defect which should hopefully get a repro produced and a fix applied soon.