NVIDIAGameWorks/PhysX #300: Happy with SDK features. But failing to link libraries to my own code in Xcode. [C++/Xcode noob]

MrMechanical, 2020-05-25:
I am a C# user, so I am unfamiliar with libraries and XCode. How do I make it so I can use the PhysX sdk in my own C++ Xcode project? I tried dragging the **Physx-4.2/physx/bin/mac.x86_64**, **Physx-4.2/physx/pxshared**,  into my frameworks and folder. I am trying my best I used to be a Unity user but I am working on integrating the SDK into my custom C++ engine. The sdk's flexiblity is very impressive so I'm excited to use this physics software. I am doing the best to follow the instructions @ https://github.com/NVIDIAGameWorks/PhysX and the specific mac instruction but Xcode keeps telling me it can't find the Px.h file or other header files.

Thank you so much for the help, I am extremely thankful for any guidance.

MrMechanical, 2020-05-25:
Is there any sort of example project folder I can copy the structure from? This might help me to understand what to drag into my frameworks folder in Xcode.  I am unfamiliar with the specific settings required for my custom XCode project. Thank you for the help.

AlesBorovicka, 2020-05-25:
The snippets are a good example I guess. They do include the PhysX SDK include directory and the required PhysX libs.
If you compile the INSTALL project of PhysX SDK, it will create a distribution package in the install folder. There should be the required libraries and includes.

MrMechanical, 2020-05-25:
Thank you so much for your help! This made my day. Which install folder should I choose?
[image]
[image]

I am an absolute Xcode noob. Will dragging this to the framework folder of my GameEngine do the trick (in terms of locating header files)? 

Thanks a lot!

AlesBorovicka, 2020-05-25:
The physx install folder should have the physx include files. 

The pxshared are just headers, use the include directory from pxshared directly.

MrMechanical, 2020-05-25:
Thank you so much for your help! Everything is much less confusing now. I apologize for my inexperience. Is /Users/myname/Documents/PhysX-4.1/pxshared/physx/include the correct search path?

MrMechanical, 2020-05-25:
I'm sorry maybe I have misunderstood the instructions. Is it intended for me to customize PhysxSDK.workspace to my needs? This would work for me.

MrMechanical, 2020-05-25:
Ok. I figured it out. I can just copy the settings used by PhysXSDK.xcodeproject.