NVIDIAGameWorks/PhysX #301: Exactly one of NDEBUG and _DEBUG needs to be defined! [new to C++][Implementing SDK into project]

MrMechanical, 2020-05-25:
I'm not sure if this is the right place to post this. I am struggling with this error in my custom C++ game engine's implementation's of Physx4.1 SDK. Do I Just define NDEBUG or am I doing something wrong? I apologize for my inexperience, I am very much a beginner when it comes to implementing libraries.

Thank you so much for the help, I am extremely grateful.

AlesBorovicka, 2020-05-26:
Check your configuration pre-processor defines, there should be just one of the defines. In your project where you include physx headers, you should have one of those defines.

Woking-34, 2020-05-28:
If you are using CMake try explicitly declaring your build type, like:
cmake .. -DCMAKE_BUILD_TYPE=Debug "-GUnix Makefiles"

mattearly, 2021-09-06:
I had to add it to the Preprocessor Definitions in Visual Studio when in Release Mode.  There may be a better way, but I could not find any other half-decent solution. 
[image]