NVIDIAGameWorks/PhysX #303: Update vertices of convex hulls without updating it.

MrMechanical, 2020-05-26:
Creating new convex hulls can be slow. Any way to directly change the vertices of the (convex) hulls?

AlesBorovicka, 2020-05-27:
No, you cannot change a vertex on a created hull. You need to recook it, if the input vertices changed. How many input vertices you have for the convex cooking?

MrMechanical, 2020-05-27:
500 vertices. I am doing a simulation at 2000hz for analysis. So it has to work under 0.5ms, is this possible? Thank you for your help.

AlesBorovicka, 2020-05-28:
Yes, it should be faster than that. Have you checked:
https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/snippets/snippetconvexmeshcreate/SnippetConvexMeshCreate.cpp
In this snippet there are various ways how to create a convex mesh with timings, so you can check the time needed to construct a hull. If insertion callback is used, you should be able to create the hull in roughly 0.3ms.

MrMechanical, 2020-05-28:
For me it took 0.6 ms on my i5 2020 macbook air. What settings/cpu are you using? Thank you for your help. If I can get 0.3ms that would be good enough for me.

AlesBorovicka, 2020-05-28:
There are different settings, make sure you use insertion callback - not call cookConvexMesh but createConvexMesh. 
If you need the cooking speed, you can increase the PxCooking::gaussMapLimit parameter to 256 and set the convex mesh desc flag  eFAST_INERTIA_COMPUTATION. You can modify the mentioned snippet to see, how the perf changes for each of these settings.

MrMechanical, 2020-05-28:
Thank you for your help. These are the flags I'm using. PxConvexFlag::eCOMPUTE_CONVEX | PxConvexFlag::eDISABLE_MESH_VALIDATION | PxConvexFlag::eFAST_INERTIA_COMPUTATION;

Is it possible to entirely disable inertia computation? All my objects are kinematic.

AlesBorovicka, 2020-05-29:
No, that is not possible. But with the createConvexMesh and insertion callback you should really not have it that slow. 
Have you also try to increase the gaussMapLimit?

MrMechanical, 2020-05-29:
I have it set at 256. Could I reuse the triangles frame to frame? The triangles aren't guaranteed to form a convex object, but I am willing to tolerate some error in the collision detection. Perhaps only computing the vertices while keeping the triangles is possible?