NVIDIAGameWorks/PhysX #304: PxD6Joint unexpected behavior with the linear drive motor

kafumanto, 2020-05-27:
Hi, I have a locked PxD6Joint, limited only on the vertical axis (Y), with a linear drive velocity target of 0 and unlimited drive force. When the local frame of the second actor is not under the center-of-mass of the actor, the drive "fails" and the actor "slips down" instead of stay still.

I would appreciate any help to understand the reason behind it. It has been reproduce on both PhysX 3.4 and 4.1. Here a complete Snippet test: https://gist.github.com/kafumanto/cbdf295cad8283101f1a781e54442850 .

So... Why the object slips down?

- it seems that it's receiving a bigger vertical force proportional to the distance of its center-of-mass to the local frame of the joint; while I would expect an angular momentum, the linear one is unexpected.
- also if the above is correct, the motor in the joint is configured with an unbound force, so it should be able to counter-act the vertical force and keep the object still. But it fails.

The setup (I'ma coder... :) ):
[image]

The unexpected result, i.e. the object slipping down (the changes in speed are caused by interactions with the drive motor [see linked demo]):
[image]

Here the relevant code for the test:
```c++
PxRigidDynamic * base = PxCreateDynamic(*gPhysics, PxTransform(PxVec3(0, 0.51f, -30)), PxBoxGeometry(0.5f, 0.5f, 0.5f), *gMaterial, 1.0f);
base->setMass(1);
base->setRigidBodyFlag(PxRigidBodyFlag::eKINEMATIC, true);
gScene->addActor(*base);

PxRigidDynamic * movable = PxCreateDynamic(*gPhysics, PxTransform(PxVec3(0, 10, -25)), PxBoxGeometry(0.5f, 0.5f, 0.5f), *gMaterial, 1.0f);
movable->setMass(1);
movable->setLinearVelocity(PxVec3(0, 0, 0));
gScene->addActor(*movable);

PxTransform jT = PxTransform(movable->getGlobalPose().p
#ifdef UNALIGNED_JOINT
    - PxVec3(0, 0, 1)
#endif
);
PxD6Joint* j = PxD6JointCreate(
	*gPhysics,
	base,
	base->getGlobalPose().getInverse().transform(jT),
	movable,
	movable->getGlobalPose().getInverse().transform(jT)
);
j->setMotion(PxD6Axis::eY, PxD6Motion::eLIMITED);
j->setLinearLimit(PxJointLinearLimit(gPhysics->getTolerancesScale(), 5.f));
j->setDrive(PxD6Drive::eY, PxD6JointDrive(0, 1000, FLT_MAX, true));
j->setDriveVelocity(PxVec3(0.f, 0.f, 0.f), PxVec3(0, 0, 0));
```

kstorey-nvidia, 2020-05-28:
Can you try increasing the number of solver iterations to see if that improves the drive's performance when the anchor location is offset?

If that does, then this is likely caused by the offset making this a harder problem because the spring forces are applied through the anchor location, causing the body to rotate. This must be counteracted by the other linear and angular constraints introduced by the D6 joint, causing a feedback loop that will converge on the result you expect, but may need additional solver iterations.

Using a PxArticulationReducedCoordinate to represent the system rather than a set of rigid bodies and joints should resolve this issue without needing to increase solver iteration counts.

However, a very small amount of velocity-level drift may still persist, especially if you extend to a more complex problem than simply 2 bodies and a single joint due to either insufficient solver iterations to converge or FP rounding errors accumulating. 

To overcome this, you would also likely need some additional error correction term to correct any accumulated positional errors over frames. This could be achieved by using the stiffness term of the spring and dynamically adjusting the positional target based on the target velocity.

kafumanto, 2020-05-28:
Hi Kier, thanks a lot for your explanation and suggestions.

Calling `setSolverIterationCounts()` on the movable actor has a limited impact: the actor still slips down, but slower than before. Following your suggestion, I played with the stiffness term of the joint drive, the result is better (also with the standard solver iteration counts). The object, after some lag, can stay still. But...

Increasing the mass of the movable actor (from 1Kg to 500Kg - a weight more realistic for the range of the objects I would need to simulate), there's no way to make the actor still or - worst - lift it up (the test snippet has also this option). Also combining both the above workarounds, the actor falls down. It seems that the drive motor is unable to produce the force required to counteract such mass. Removing the offset between the joint and the movable actor, the drive can then control again easily the actor (also with a mass of 500'000Kg). There are other options I can try?

Unfortunately I can't use PxArticulationReducedCoordinate at the moment, the project I'm working on is on UE4 and so I'm limited to PhysX 3.4. Do you know if - from a technical point of view - is possible to port the  PxArticulationReducedCoordinate feature to PhysX 3.4 as simple "_client code_" (i.e. without the need of re-compiling the PhysX lib shipped with UE4)? In this case, I could evaluate to work on the porting.

Thanks a lot for your help.

kstorey-nvidia, 2020-05-28:
When you increase the actor mass, by how much do you increase the stiffness and damping terms?

In theory, if you can get results with a 1kg box, you should be able to get the same results with a 500,000kg box, it's just that the stiffness and damping terms required to achieve that will be much larger.

There are maximal coordinate articulations in PhysX 3, but they wouldn't be suitable for what you are trying to do because they don't support prismatic joints. To get an articulation implementation that is suitable for your use-case, you would need to compile UE4 against PhysX 4 and then write some custom code to instantiate the articulations somehow.

This would be quite an involved process. However, maybe this might help you?

https://docs.nvidia.com/isaac/isaac_sim/setup.html

This is NVIDIA's Isaac Sim project, which is a robotics simulation platform using a branch of UE4 using PhysX 4. It's based on an older version of UE4.

kafumanto, 2020-05-28:
Without the offending "offset" and a mass of 500,000 Kg, stiffness and damping set to 1,000 are enough to move the object. But with the "offset", I tried values of 5,000 - 1,000,000 - 10,000,000 - 100,000,000 and the results are all the same, with the object falling.

Thanks a lot for the link, it could be helpful in the future. At the moment I don't think we'll be able to switch to a custom build of UE4, it would make the internal production too complex for our small team (while porting the relevant code would allow us to continue using the standard UE4 shipped by Epic - but I understand that it could not be feasible).