NVIDIAGameWorks/PhysX #306: PVD crashes when there are joints in the scene

jankrassnigg, 2020-06-04:
PVD version: 3.2019.04.26214843 (latest as far as I know)
PhysX version: 4.1

I have a scene that contains only a joint (I tested spherical, fixed, 6dof, prismatic, distance) and an actor. The joint is connected to one (or two) dynamic actors (in case of 1, the other connection is just null). The actor just uses a box shape.

If I now connect to the PVD, it crashes with:
D:\Code\PhysX\physx\source\pvd\src\PxPvdDataStream.cpp(508) : Assertion failed: success

This happens during the call to PxPvd::connect().

If I first connect to the PVD with a working scene (no joint) and then add a joint later, it crashes with the same assert, just called through the joint creation function, e.g. PxPrismaticJointCreate().

Callstack:

[image]

This only happens when the PVD connection uses PxPvdInstrumentationFlag::eDEBUG, if that flag is not set, everything works fine (but then the PVD is pretty useless).

Otherwise I can use PVD with complex scenes with all sorts of shapes, triggers, etc., but not with any single joint. Without the PVD the joints work as expected, so I would be surprised if there is any error in how they are set up.

kstorey-nvidia, 2020-06-05:
Did you call:

PX_C_EXPORT bool PX_CALL_CONV PxInitExtensions(physx::PxPhysics& physics, physx::PxPvd* pvd);

This call initializes PhysX extensions. One of the main jobs of this call is to register all the types in extensions with PVD.

jankrassnigg, 2020-06-05:
That was indeed the missing piece, thanks!

closed 2020-06-05 by jankrassnigg

morphogencc, 2021-06-10:
I'm running into this same issue -- I can use PVD without a joint, or I can run the program with my joint without PVD, but not both.  Where should I make the call that @kstorey-nvidia mentioned?  I added it into the header file where I include PhysX, but that doesn't seem to resolve the issue.  I also included `extensions/PxExtensionsAPI.h` which seems to have this call as well.

kstorey-nvidia, 2021-06-10:
Did you call PxInitExtensions, passing the pointer to your PVD instance to it? 

An example of joints working with PVD can be found in SnippetJoint.cpp. See the initPhysics(...) method.

The crucial requirement is that you have to call this method:

PxInitExtensions(*gPhysics, gPvd);

You can use joints without PVD without calling this method, but joints are part of the PhysX extensions package. The PVD system is completely generic so the call to this init function registers the metadata for all the types in extensions with PVD. Once you have done this, the PVD instance should not crash with a scene with joints.

morphogencc, 2021-06-11:
That was it -- thank you!

pkxpp, 2021-06-28:
> Did you call:
> 
> PX_C_EXPORT bool PX_CALL_CONV PxInitExtensions(physx::PxPhysics& physics, physx::PxPvd* pvd);
> 
> This call initializes PhysX extensions. One of the main jobs of this call is to register all the types in extensions with PVD. This includes the joint types that are in extensions.

This is useful for me.