NVIDIAGameWorks/PhysX #309: Are we going to see another update at some point?

Zeblote, 2020-06-13:
Hey, so it's been almost a year since PhysX 4.1, and complete radio silence since then. I'm sure lots of optimizations and bug fixes have added up over this time. Wouldn't it be nice to see another update?

Hoikas, 2020-07-11:
It would be nice if pull requests that have been accepted into NVIDIA's internal main were also merged here on GitHub. As it stands, they are shunted off into the "closed" tab where it is difficult to find them.

Zeblote, 2020-07-25:
https://twitter.com/PierreTerdiman/status/1284124994751037442 PhysX 5.1 seems to be a thing already?

amoravanszky, 2020-07-29:
The PhysX 5.1 SDK is not yet released, it is under development internally, and Pierre is clearly too excited to keep quiet about it.  

PhysX 5 however is already being used in NVIDIA Omniverse, and in particular it is going out to all developers who have registered to to access Isaac Sim: https://developer.nvidia.com/isaac-sim

We do plan to update this github with a source release eventually, but I don't have a date for when that will happen.  Since we are now developing a full stack of our own tooling to go on top of the SDK, as well as features well beyond just rigid bodies, finishing it is taking longer than in the past when the scope was limited to a game physics runtime library.

Zeblote, 2020-08-02:
Hi @amoravanszky, thanks for responding! Am I understanding it correctly that registering to access Isaac Sim does **not** give access to the PhysX 5 SDK (with a license to redistribute the binaries)?

amoravanszky, 2020-08-05:
Correct, you can't develop your own software using it, it is only for using as part of Isaac Sim.  So, yes, a C++ software development kit of PhysX 5 for making your own software is not yet available.

Zeblote, 2020-10-02:
Any news though? 2020 is going to be over soon!

Dasein123, 2020-10-23:
I'm also curious tbh

Dasein123, 2020-12-03:
Would it be reasonable to guess that it might be 2021 source release instead of 2020 due to it being end of the year?

Zeblote, 2020-12-08:
Maybe 2022 is more likely?

dyanikoglu, 2021-02-01:
Maybe 2023? 🙄

bryansee, 2021-03-03:
Maybe sometime after the end of the coronavirus pandemic, which happens at the heat death of the universe or the end of time, perhaps?

Zeblote, 2021-03-03:
3 months to go until the first anniversary of this issue!

dyanikoglu, 2021-03-04:
Unreal's Chaos becoming stable vs. Nvidia releasing PhysX 5. Who would win? :)

Daasin, 2021-12-15:
> The PhysX 5.1 SDK is not yet released, it is under development internally, and Pierre is clearly too excited to keep quiet about it.
>
> PhysX 5 however is already being used in NVIDIA Omniverse, and [...]
 
---
 
I know recent events could've delayed things but was previewed in 2019 & it's almost 2022 now. 

Given that it's shown to be complete enough in Omniverse, would it not be fair for those who were led to believe that they could depend on that release to atleast get a public announcement saying it's not coming?  @amoravanszky as opposed to people searching and seeing the articles that PhysX5 with DEM would be available as Open-Source? 

Surely you can understand the frustrations & disappointment people may feel after plans would've been made or whatever else would have been impacted by the radio-silence and feeling as if it's been brushed away and left out to dry 🙁