NVIDIAGameWorks/PhysX #311: PGS vs TGS

rouxfrancis, 2020-06-15:
Hi, I have a bug with the TGS solver. On the following video I simulate the same scene with the two types of solvers and I get totally different results:

https://photos.app.goo.gl/EZy176ogMoAzQpLM7

FYI I implemented a grabber which can dynamically create a PxSphericalJoint (red dot) or a PxFixedJoint (red square) to move the rigid objects that have been raycasted. Any suggestion is welcome to overcome this problem.

kstorey-nvidia, 2020-06-16:
Would you be able to provide a PVD capture of this configuration so we can investigate what might be going wrong?

rouxfrancis, 2020-06-16:
It may take a little time, I did my own integration of PhysX in Unity but I have not yet managed the PVD feature. I will contact you as soon as I have a capture of the scene.

rouxfrancis, 2020-06-17:
While investigating I found a bug which changed the situation a little, but I still have problems with the TGS solver:
[PGS vs TGS.zip](https://github.com/NVIDIAGameWorks/PhysX/files/4794962/PGS.vs.TGS.zip)