NVIDIAGameWorks/PhysX #312: Simulating vehicles connected by joint

zemlifr, 2020-06-16:
Hi, I am trying to simulate trucks with PxVehicleNoDrive and spherical joints.
I tried to extend SnippetVehicleNoDrive by adding second vehicle and connecting them with joint, but it doesn't work for me.

When I apply torque to 1st vehicle to move forward, it doesn't move, only wheels rotate rapidly. second vehicle doesn't move at all. This happens only if both vehicles are stationary when I apply torque. When they both have some velocity, i am able to pull 2nd vehicle with first.

Tried with Fixed, Revolute and Spherical joint.

Setup code (from snippet):

```
  VehicleDesc vehicleDesc = initVehicleDesc();
  gVehicleNoDrive = createVehicleNoDrive(vehicleDesc, gPhysics, gCooking);
  PxTransform startTransform(PxVec3(0, (vehicleDesc.chassisDims.y*0.5f + 
  vehicleDesc.wheelRadius + 2.0f), 0), PxQuat(PxIdentity));
  gVehicleNoDrive->getRigidDynamicActor()->setGlobalPose(startTransform);
  gScene->addActor(*gVehicleNoDrive->getRigidDynamicActor());

  VehicleDesc vehicleDesc2 = initVehicleDesc();
  gTowedVehicle = createVehicleNoDrive(vehicleDesc2, gPhysics, gCooking);
  PxTransform startTransform2(PxVec3(0, (vehicleDesc2.chassisDims.y*0.5f + 
  vehicleDesc2.wheelRadius + 2.0f), -10.f), PxQuat(PxIdentity));
  gTowedVehicle->getRigidDynamicActor()->setGlobalPose(startTransform2);
  gScene->addActor(*gTowedVehicle->getRigidDynamicActor());

  //Set the vehicle to rest in first gear.
  //Set the vehicle to use auto-gears.
  gVehicleNoDrive->setToRestState();
  gTowedVehicle->setToRestState();

  PxTransform jointTransform(PxVec3(0, (vehicleDesc2.chassisDims.y*0.5f + 
vehicleDesc2.wheelRadius + 1.0f), -5.5f), PxQuat(PxIdentity));
  PxTransform jointTransform2(PxVec3(0, (vehicleDesc2.chassisDims.y*0.5f + vehicleDesc2.wheelRadius + 1.0f), 5.5f), PxQuat(PxIdentity));

PxSphericalJointCreate(*gPhysics, gVehicleNoDrive->getRigidDynamicActor(), jointTransform, gTowedVehicle->getRigidDynamicActor(), jointTransform2);
```

gyeomannvidia, 2020-06-17:
At very low speed and in the absence of throttle or drive torque, the car is held in place with a velocity constraint at each tyre contact patch.  This asymptotically brings the car to rest and holds the car at rest until throttle is applied. 

Can you try applying a very small drive torque to the front wheels?  This ought to free the velocity constraints on the rigid body, leaving the rigid body free to respond to user-applied torques.

Cheers,

Gordon

zemlifr, 2020-06-25:
Thank you for your reply. Applying small torque to second vehicle works ok.

closed 2020-06-25 by zemlifr