NVIDIAGameWorks/PhysX #313: physx4.1 mingw compile android fail

wangtianhang, 2020-06-28:
sorry for my poor english.
after i generate android makefile, i enter the folder android-arm64-v8a-release and try to make.
but seemly it enter cmd.exe and nothing happen.

my cmake is 3.17 python is 2.7.6

wangtianhang, 2020-06-28:
[https://github.com/wangtianhang/Misc/blob/master/physx_make_error.png](url)

wangtianhang, 2020-06-28:
[https://github.com/wangtianhang/Misc/blob/master/physx_cmake_success.png](url)

wangtianhang, 2020-06-28:
this is the cmake generate project~
[https://github.com/wangtianhang/Misc/blob/master/android-arm64-v8a-release.zip](url)

wangtianhang, 2020-06-28:
i think my environment is ok, because i can compile phyx3.4 android

AlesBorovicka, 2020-06-29:
It should create MinGW makefiles. Have you tried the MinGW make? 
https://osdn.net/projects/mingw/releases/

wangtianhang, 2020-06-29:
well,finally,i found it shoud use mingw32-make rather than only make... and i use it in cmd environment.
the document only said use make. =.=