NVIDIAGameWorks/PhysX #320: When does PxContactModifyCallback onContactModify get raised

ronbrogan, 2020-07-08:
I'm working on integrating PhysX into my engine, and I'm a bit confused as to when to use PxContactModifyCallback.
My assumption was that whenever any meaningful contact is determined between Actors, this callback will be invoked. However, this is not what I experience. 

I have a PxRigidBody that slides around some terrain (PxRigidStatic) with 0 friction. Hundreds of frames of sliding on this terrain can go by without this event being raised, but then it (seemingly) randomly will be invoked. 

It seems that I'm likely more interested in PxSimulationEventCallback's onContact callback, but I would like to understand the distinction.

kstorey-nvidia, 2020-07-08:
Contact modify callback (onContactModify) should be called for every contact pair that generated contacts that requested contact modification in the filter shader. It will not be called unless the pair requested it because it adds overhead to the simulation (both simulation cost and memory cost). If it isn't being called when a pair of shapes are generating contacts, that should be because the pair didn't raise PxPairFlag::eMODIFY_CONTACTS.

The frequency that onContact callback is called for a pair varies depending on the flags involved - if you request only TOUCH_FOUND and TOUCH_LOST flags, it will be called only when the pair first starts/ends touching. If you request TOUCH_PERSIST, it will get called every frame the shapes are touching provided they are awake and simulating.

ronbrogan, 2020-07-08:
Thanks for the quick response here, that makes a lot of sense. 
I'm currently using a filter shader that always sets eMODIFY_CONTACTS, so I must be doing something wrong on my end to see this behavior. 
Thanks again for the concise explanation!

closed 2020-07-08 by ronbrogan