NVIDIAGameWorks/PhysX #321: Error in compiling with clang 10.0

keto33, 2020-07-09:
I tried to compile PhysX with clang 10.0 on Ubuntu 20.04, but it was stopped by the error:

```
PxProfileEventBufferAtomic.h:293:5: error: use of function '__builtin_alloca' is discouraged; there is no way to check for failure but failure may still occur, resulting in a possibly exploitable security vulnerability [-Werror,-Walloca]
                                PX_ALLOCA(tempBuffer, uint8_t, sizeToWrite);
                                ^
```

AlesBorovicka, 2020-07-21:
Hi,
 as a temp workaround you can disable that warning in the cmake file:
https://github.com/NVIDIAGameWorks/PhysX/blob/4.1/physx/source/compiler/cmake/linux/CMakeLists.txt
Sorry for the inconvenience.

keto33, 2020-07-21:
I just opened this issue for your reference. I did successfully compile using clang 9.0.

hojalot, 2020-08-30:
The specific change I made to workaround this was adding -Wno-alloca to PHYSX_CXX_FLAGS ~line 36 in physx/source/compiler/cmake/linux/CMakeLists.txt

ayoub-belarbi, 2021-04-22:
Thank you for reporting this issue! We just released PhysX 4.1.2 that fixes this issue (along with other fixes). Make sure to check it out :)
Your contribution is greatly appreciated!

closed 2021-04-22 by ayoub-belarbi

janinurminen, 2021-07-20:
Issue is still present in macOS build, just took latest (versio 4.1 codes from this repo).