NVIDIAGameWorks/PhysX #323: HelloGRB segmentation fault

grizzi, 2020-07-17:
Hi, 
I have been trying Physx with the goal of using GPU accelerated dynamics. The only snippet using gpu `HelloGRB_64` segfault. I have turned on my GPU . My system specs are:
- GPU: `GeForce MX150`
- Driver version: `440.100`

I have built physx with the `linux` preset in release mode. Do you know what can go wrong here?
Cheers,
Giuseppe

kstorey-nvidia, 2020-07-20:
Could you post a callstack please? Could you also try this with debug mode to see if you get more information? Has the application output any information prior to the segfault? If there's a problem, PhysX should usually send out a number of errors to the error stream.

grizzi, 2020-07-21:
I have tried the debug mode and I get only the following output:
`Segmentation fault (core dumped)`

The visualization window tries to open and then the segfault happen. Changing to use cpu instead of gpu fixes this so my guess is something gpu related. All other snippets work ok. I tried to run in gdb and this is the call stack:

```
Thread 1 "SnippetHelloGRB" received signal SIGSEGV, Segmentation fault.
0x0000000000554f84 in physx::Sc::Scene::Scene (this=0x14ef1e0, desc=..., contextID=21950912)
    at /home/giuseppe/git/PhysX/physx/source/simulationcontroller/src/ScScene.cpp:728
728			mGpuWranglerManagers = PxvGetPhysXGpu(true)->createGpuKernelWranglerManager(mLLContext->getCudaContextManager(), getFoundation().getErrorCallback(),desc.gpuComputeVersion);
(gdb) bt
#0  0x0000000000554f84 in physx::Sc::Scene::Scene (this=0x14ef1e0, desc=..., contextID=21950912)
    at /home/giuseppe/git/PhysX/physx/source/simulationcontroller/src/ScScene.cpp:728
#1  0x00000000004711c1 in physx::Scb::Scene::Scene (this=0x14ef1d0, desc=..., contextID=21950912)
    at /home/giuseppe/git/PhysX/physx/source/physx/src/buffering/ScbScene.cpp:344
#2  0x0000000000430e9e in physx::NpSceneQueries::NpSceneQueries (this=0x14ef1c0, desc=...)
    at /home/giuseppe/git/PhysX/physx/source/physx/src/NpScene.cpp:88
#3  0x000000000043118a in physx::NpScene::NpScene (this=0x14ef1c0, desc=...)
    at /home/giuseppe/git/PhysX/physx/source/physx/src/NpScene.cpp:106
#4  0x000000000042a441 in physx::NpPhysics::createScene (this=0xe60c70, desc=...)
    at /home/giuseppe/git/PhysX/physx/source/physx/src/NpPhysics.cpp:269
#5  0x0000000000414ecc in initPhysics ()
    at /home/giuseppe/git/PhysX/physx/snippets/snippethellogrb/SnippetHelloGRB.cpp:135
#6  0x000000000041626c in renderLoop ()
    at /home/giuseppe/git/PhysX/physx/snippets/snippethellogrb/SnippetHelloGRBRender.cpp:117
#7  0x0000000000415964 in snippetMain ()
    at /home/giuseppe/git/PhysX/physx/snippets/snippethellogrb/SnippetHelloGRB.cpp:199
#8  0x00000000004145a2 in main (argc=1, argv=0x7fffffffa2b8)
    at /home/giuseppe/git/PhysX/physx/snippets/snippetcommon/ClassicMain.cpp:34
```

kstorey-nvidia, 2020-07-21:
Thanks for the details. Could you please let me know which version of Linux you are using?

This looks like it is failing to initialize the GPU pipeline, which might suggest that the application is maybe using your integrated GPU rather than the MX150.

Normally, if there is not a compatible GPU, driver etc., this should result in PxCreateCudaContextManager returning null, which triggers us to fall back to the CPU pipeline. It seems like that must not be the case with your machine so it's proceeding like there is a compatible GPU but then crashing later.

Could you please confirm whether the following line:

gCudaContextManager = PxCreateCudaContextManager(*gFoundation, cudaContextManagerDesc, PxGetProfilerCallback());

returns a valid pointer? Can you inspect this pointer? Can you call an arbitrary method on it? I presume that must be possible because the code in the snippet calls the validation check.

Following that, could you please verify whether:

PxvGetPhysXGpu(true)

returns a valid or nullptr in that initialization code? The callstack looks like PxvGetPhysXGpu might be returning null or garbage?

grizzi, 2020-07-21:
thanks for the fast reply! I added some prints and this is what I get:
```
GPU context manager pointer: 0x3056450
gCudaContextManager->contextIsValid() returned true
PxvGetPhysXGpu(true) returned: 0x3547b50
sceneDesc.gpuComputeVersion returned: 0
Setting cuda context manager
Creating scene
Segmentation fault (core dumped)

```
I think it is detecting the gpu because when turning nvidia gpu off,  I have some warnings printed to console that inform code is falling back to cpu as expected. This is an extract of the modified code with printouts
```
  gCudaContextManager = PxCreateCudaContextManager(*gFoundation, cudaContextManagerDesc, PxGetProfilerCallback());	//Create the CUDA context manager, required for GRB to dispatch CUDA kernels.
  std::cout << "GPU context manager pointer: " << gCudaContextManager << std::endl;

  if( gCudaContextManager )
  {
    if( !gCudaContextManager->contextIsValid() )
    {
      std::cout << "gCudaContextManager->contextIsValid() returned false" << std::endl;
      gCudaContextManager->release();
      gCudaContextManager = NULL;
    }
    else
      std::cout << "gCudaContextManager->contextIsValid() returned true" << std::endl;
  }
  PxSceneDesc sceneDesc(gPhysics->getTolerancesScale());
  // added to  check the output pointer is valid
  PxPhysXGpu* physxGpu = PxvGetPhysXGpu(true);
  std::cout << "PxvGetPhysXGpu(true) returned: " << physxGpu << std::endl;
  std::cout << "sceneDesc.gpuComputeVersion returned: " << sceneDesc.gpuComputeVersion << std::endl;
```

My os is `Ubuntu18.04LTS` and the kernel version is `5.3.0-62-generic`.

kstorey-nvidia, 2020-07-21:
Is this a 64-bit or 32-bit config? Asking because the pointers look like they could be just 32-bit.

grizzi, 2020-07-21:
It is 64bit

grizzi, 2020-07-22:
@kstorey-nvidia  Any idea of what can go wrong here?

kstorey-nvidia, 2020-07-22:
Not currently, I'm afraid. The point where it is failing is when it is trying to load/cache the GPU kernels from the PhysXGPU.so so they can be executed during simulation. At this stage, there may be some JIT compilation inside the driver taking place to target your specific GPU.

In order to get this far, CUDA had to start up OK and find a compatible GPU (as you noted, disabling your GPU results in PhysX detecting no compatible GPU and falling back on CPU).

As I understand, the MX150 is a 4GB Pascal-based notebook GPU. That should be sufficient to run GPU PhysX, but I had to admit that I don't think we in the team currently have access to a system with that GPU to test locally, although we do have GPUs using the same architecture, which have been confirmed to work with Ubuntu 18.04LTS.

Just as a follow-up to what you said about "it is 64-bit" - this means you are definitely building the "Linux64" config? 32-bit support was removed in CUDA 10 on Linux so there may be issues trying to use a Linux32 config.

grizzi, 2020-07-22:
Thanks for your answer! Yes, the build is for linux 64. I would be happy to hear some news if you find the reason for the issue. I will do the same if I manage, for some reason. to run the simulation on GPU. In the meanwhile thanks a lot