NVIDIAGameWorks/PhysX #325: Static CUDA libraries

tysoni, 2020-07-24:
I understand PhysX CUDA code is not public, but is there any possibility that static libraries could be released so that PhysXGpu_64.dll and PhysXDevice64.dll don't need to be bundled with applications?

kstorey-nvidia, 2020-07-30:
If you aren't using the GPU features, you don't need to include those DLLs. They are only required if you make use of GPU features.

Currently, PhysXGpu is only shipped as a DLL for a number of reasons. The main reason is compatibility - we can use the same GPU DLL between different VS versions, including future versions. If we shipped as a static library, we would need to ship libs built for all variations of build tools and also for configs like /MT vs /MD. The size of the distro would become much larger as a result. 

It is technically possible to support a static lib but there are practicality challenges.

morphogencc, 2021-02-11:
I would be interested in a static library as well -- it'd make my use cases much more straightforward.

hadar-segal-isi, 2024-11-21:
Thanks for explaining to problem related to shipping static libs. Any reason why the code for PhysXGpu_64.dll is not public?