NVIDIAGameWorks/PhysX #327: I cannot set the tessellation flag of Height Fields.

k-sonobe, 2020-07-28:
Regardless of whether the tessellation flag is set or not(PxHeightFieldSample :: setTessFlag() or PxHeightFieldSample :: clearTessFlag()), the triangulation is as shown in the figure below.

```
for (PxU32 i = 0; i < col_num_; i++)
{
	for (PxU32 j = 0; j < row_num_; j++)
	{
		samples[i * row_num_ + j].height = (PxI16)((rand() % (rand_num + 1)));
		samples[i * row_num_ + j].clearTessFlag();   //or samples[i * row_num_ + j].setTessFlag();
		samples[i * row_num_ + j].materialIndex0 = 1;
		samples[i * row_num_ + j].materialIndex1 = 1;
	}
}
```

[image]

[image]

AlesBorovicka, 2020-07-28:
The tessellation flag is stored in materialIndex0, so I think you are overwriting that setting in your code snippet, can you try to swap the lines:
`		
                samples[i * row_num_ + j].materialIndex0 = 1;
                samples[i * row_num_ + j].materialIndex1 = 1;
                samples[i * row_num_ + j].clearTessFlag();   //or samples[i * row_num_ + j].setTessFlag();
`

k-sonobe, 2020-07-28:
Thank you for the very fast response and helpful information.

However, as you pointed out, rewriting the code did not change the result.
Are there any other mistakes?