NVIDIAGameWorks/PhysX #328: Compilation error and warnings (gcc 10.1) . nonnull, restrict and class-memaccess

phcerdan, 2020-08-08:
```
PhysX/src/physx/../pxshared/include/foundation/unix/PxUnixIntrinsics.h:162:15: error: argument 1 null where non-null expected [-Werror=nonnull]
```

```
PhysX/src/physx/source/geomutils/src/mesh/GuBV4Build.cpp:298:24: error: passing argument 1 to ‘restrict’-qualified parameter aliases with argument 5 [-Werror=restrict]
```
Both can be silenced with:  `-Wno-error=nonnull -Wno-error=restrict

Another:
```
PhysX/src/physx/source/lowleveldynamics/include/DyFeatherstoneArticulationUtils.h:53:70: error: ‘void* memset(void*, int, size_t)’ clearing an object of type ‘struct physx::Cm::UnAlignedSpatialVector’ with no trivial copy-assignment; use assignment or value-initialization instead [-Werror=class-memaccess]
```

damian-tomczak, 2022-06-12:
I can't get rid of treating the warning as an error. Can anyone help?

[image]

For example, I tried to get rid of the error by "resetting" compiler flags:
`set(CMAKE_CXX_FLAGS "")
message(INFO ${PHYSX_PATH})
add_subdirectory(${PHYSX_PATH}/compiler/public externals/physx)
target_link_libraries(${PROJECT_NAME} ${PHYSX_LIBRARIES})
`

phcerdan, 2022-06-12:
Why "resetting" would solve this problem?
Try:
```cmake
set(CMAKE_CXX_FLAGS ${CMAKE_CXX_FLAGS} "-Wno-error=nonnull -Wno-error=restrict")
```

damian-tomczak, 2022-06-12:
Unfortunately it did not help:
[image]

damian-tomczak, 2022-06-12:
```
string(REPLACE " -Werror" "" CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS}")
string(REPLACE "-Werror=class-memaccess" "" CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS}")
add_subdirectory(${PHYSX_PATH}/compiler/public externals/physx)
string(REPLACE "-Werror=class-memaccess" "" CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS}")
string(REPLACE " -Werror" "" CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS}")
```