NVIDIAGameWorks/PhysX #329: Joint listener ?

rouxfrancis, 2020-08-24:
Is there a way to know the actions (force and torque) that apply to a joint? For example a joint listener, which would provide this information after each simulation step using a callback. It would be useful to determine the intensity of a collision, or to choose to break a specific joint.

kstorey-nvidia, 2020-08-24:
You can call getForce on PxConstraint and it will give the force and torque applied by the constraint.

There is also an automatic joint breaking feature in PhysX, where you define a break threshold and the joint will be automatically broken when the force required exceeds that threshold.

rouxfrancis, 2020-08-24:
I had missed this info, thank you for this quick response.

rouxfrancis, 2020-08-24:
Sorry, I closed this issue a bit quickly. By focusing more precisely on collisions I didn't see how to recover the constraint associated with the contact, to know the force of the impact.