NVIDIAGameWorks/PhysX #334: Vehicle Suspension vs Dynamics Bodies.

DaveGravel, 2020-09-09:
Hi everyone, 
I don't think my problem is a real issue, Maybe it is only a configuration problem.
I have success to fix pretty much all my old problems in this post.
https://github.com/NVIDIAGameWorks/PhysX/issues/150

Now I have the vehicle going pretty good but I have a big suspension problem.
Here a exemple: https://www.youtube.com/watch?v=K57g-AoJ-Kw

When the vehicle roll over dynamics objects I lose the suspension spring effect.
What can cause this problem, How I can fix it.
I have test with simple raycast or with sweep and I get the same result.
I have try some different query filter blocking or noblocking on the wheels but again it is the same result.

Someone know why I get this problem and how I can fix this ?
Thanks, Sorry about my english.

DaveGravel, 2020-09-15:
I get similar problem too with the d6joint and when the GPU mode is enabled.
I have create a multi-body vehicle and I use d6joint for the wheels. 
It work pretty good but when the wheels become closer to multiple dynamics objects the d6joints totally loses the suspensions.
This problem don't happen when I use the CPU mode.
You can see the problem here in this video.
At around 3min45 I change for use the CPU mode and you can see the difference with the suspenssions.
https://youtu.be/JynRY9QNEdw

kstorey-nvidia, 2020-09-15:
You may find that setting gpuMaxNumPartitions in PxSceneDesc to 32 instead of its default 8 helps this problem. This number must be a power of two value between 1-32 (32 is the max). If this doesn't resolve the issue, you may also need to increase the solver iteration count.

The problem is caused by how the solver handles overflowing the maximum number of attachments. It ends up falling back on a solution that converges worse.

There is an improvement to the GPU solver that should completely resolve issues like this coming up in PhysX 5, where a large chunk of the GPU solver has been rewritten.

DaveGravel, 2020-09-15:
Amazing thanks a lot for the tip.
I need to make some more adjustments in my scene but it look to work a lot better now.
https://www.youtube.com/watch?v=Dbg4kffAXOo

It don't look to fix the problem with the physx vehicle component sweep wheels.
The problem look similar but I think it coming from a bad configuration I need to test more with this one.

kstorey-nvidia, 2020-09-15:
I'm glad that this helped!

"It don't look to fix the problem with the physx vehicle component sweep wheels."

The vehicle component wheels are solved externally from the rigid body solver so all the solver sees are external forces and torques being applied on the chassis body. If you use the vehicle component, you don't need to worry about many contacts affecting what the behavior of the wheels. The problem only occurs if we try to simulate wheels using constraints, and it should be a fixed issue in the next major PhysX release. For now, increasing the gpuMaxNumPartitions will not completely solve the issue, but it will make reduce the artefacts.

DaveGravel, 2020-09-15:
Thanks a lot again I very appreciate your help.