NVIDIAGameWorks/PhysX #335: Changing the scale of trianglemesh shape after creation

benoitjacquier, 2020-09-13:
Hi,
I would like the change the scale of a triangle mesh / shape after it's creation
Do I need re-create the shape ?
( https://gameworksdocs.nvidia.com/PhysX/4.0/documentation/PhysXGuide/Manual/Geometry.html#mesh-scaling ) 
Or can I just change it directly by getting / setting the same geom ?
```cpp
PxGeometryHolder holder = gTriShape->getGeometry();
float s = 0.3;
holder.triangleMesh().scale.scale = PxVec3( s, s, s );
gTriShape->setGeometry( holder.any() );
```
Thanks

benoitjacquier, 2020-09-13:
Note that I need to acquire/release a reference around the NpShape::setGeometry else I have a crash
```cpp
PxGeometryHolder holder = gTriShape->getGeometry();
holder.triangleMesh().triangleMesh->acquireReference();
float s = 0.3;
holder.triangleMesh().scale.scale = PxVec3( s, s, s );
gTriShape->setGeometry( holder.any() );
holder.triangleMesh().triangleMesh->release();
```

I guess this is due to NpShape::setGeometry calling decMeshRefCount before incMeshRefCount

pkxpp, 2021-06-05:
I doubt why PxTransform has no scale member all the times~