NVIDIAGameWorks/PhysX #338: Physx 5.0 (+FEM)

Dasein123, 2020-09-15:
https://gpuopen.com/femfx/ 

https://www.techradar.com/news/nvidia-reveals-physx-50-but-amds-explosive-femfx-demos-already-stole-its-physics-thunder - Wasn't code for Physx 5 meant to be released? Even if we sort it out ourselves?

Seems to be dead and only see 4/4.1 here

alanjfs, 2020-09-16:
Haha, what is this? xD

Dasein123, 2020-09-16:
a question, i dont know where else to find the answer

alanjfs, 2020-09-18:
Looks like a question, rolled into some negative assertions, a competing product and a clickbait news article.

Perhaps the question you meant to ask was simply "When is PhysX 5.0 released?". To which I would also very much like to know the answer, because it looks amazing! Huge props to the team on this one.

Dasein123, 2020-09-18:
Yeah, sorry I was just confused about why nobody talks about it now. Wasn't it 2020 release?

kstorey-nvidia, 2020-09-18:
PhysX 5 is still in development and has not been released yet. It is, however, use inside NVIDIA in several projects.

https://github.com/NVIDIAGameWorks/PhysX/issues/309#issuecomment-665751081

Dasein123, 2020-09-18:
Thank you so much for the response! @kstorey-nvidia 

Sorry if I came off wrong in the other message 😅 I didn't mean offense, honest

Dasein123, 2020-11-20:
Closed due to not being able to know anything on the matter till official release

closed 2020-11-20 by Dasein123

Daasin, 2021-12-15:
Would this need re-opening to ask about any maximum timeframe? 🤔