NVIDIAGameWorks/PhysX #340: [Question] Is it possible to make a down-facing HeightField?

BeeMickSee, 2020-09-21:
Hello! I'm integrating PhysX 4.1 into a game engine in which -Y is up. 

After seeing the spherical gravity demo and reading that gravity is indeed functionally the same as any other force, I set the Y component of the gravity vector to a positive value and moved on. Everything was going peachy keen, until I got to HeightFields, which treat positive Y as up, and I can't find a way around it!

I thought maybe I could set the heightScale of PxHeightFieldGeometry to -1.0f, but heightScale must be positive.

I've Googled, read the docs, and even asked Twitter, but to no avail. Is there a workaround for this? Thank you!

kstorey-nvidia, 2020-09-21:
All shapes have a transform, so you can rotate the shape such that the shape's local Y axis points downwards in world space.

For this to match your source art, you would need to rotate and also mirror your heightmap sample values. This should yield the same results as having mirrored along the y-axis.

BeeMickSee, 2020-09-21:
Oh my gosh, is it really that simple?! 

I thought HeightFields only represented height values, but I guess if it's handling collisions with the same code as triangle meshes under the hood, why not allow them to rotate?

Thank you for the super quick response! I really appreciate it.

closed 2020-09-21 by BeeMickSee